AltBeacon/android-beacon-library #881: Background Monitoring using RegionBootstrap is not calling didExitRegion

mohand1993, 2019-05-23:
While monitoring a region in the Background and leaving the region after it's been detected, didExitRegion is not triggered because the scanning gets stopped when the region became in range.
### Expected behavior
Scanning should keep running even when a region is in range so that other regions and leaving the range can still be detected
### Actual behavior
Scanning stops once a range is detected
### Steps to reproduce this behavior
- Start the reference app
- Send the app to background
- Enter beacon range
<< Scanning stops>> 
- Leave the range
<<didExitRegion doesn't get triggered>> 
### Mobile device model and OS version
- Pixel 3 running Android Q-beta3
- Pixel 2 running Android P
### Android Beacon Library version
2.16.1
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-05-23:
Be sure you wait for up to 15 minutes for a region exit -- it can take that long due to Android 8+ limits on how often background jobs can run.

[image]

Here is a log excerpt showing your steps working properly on the Google Pixel 3a with Android 9:

```
05-23 09:12:19.326 31228 31228 I beaconreferenc: Late-enabling -Xcheck:jni
05-23 09:12:19.329  1235  1257 I ActivityManager: Start proc 31228:org.altbeacon.beaconreference/u0a238 for activity org.altbeacon.beaconreference/.MonitoringActivity
05-23 09:12:19.420 31228 31238 E beaconreferenc: Failed to send DDMS packet REAQ to debugger (-1 of 20): Broken pipe
05-23 09:12:19.557 31228 31228 W beaconreferenc: JIT profile information will not be recorded: profile file does not exits.
05-23 09:12:19.560 31228 31228 I chatty  : uid=10238(org.altbeacon.beaconreference) identical 10 lines
05-23 09:12:19.560 31228 31228 W beaconreferenc: JIT profile information will not be recorded: profile file does not exits.
05-23 09:12:19.582 31228 31228 I InstantRun: starting instant run server: is main process
05-23 09:12:19.592 31228 31228 I BeaconManager: BeaconManager started up on pid 31228 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
05-23 09:12:19.598 31228 31228 D BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-23 09:12:19.600 31228 31228 D BeaconReferenceApp: setting up background monitoring for beacons and power saving
05-23 09:12:19.687 31228 31228 I ScanJob : Using immediateScanJobId from manifest: 208352939
05-23 09:12:19.690 31228 31228 I ScanJob : Using periodicScanJobId from manifest: 208352940
05-23 09:12:19.690 31228 31228 W JobInfo : Requested interval +5m0s0ms for job 208352940 is too small; raising to +15m0s0ms
05-23 09:12:19.691 31228 31228 W JobInfo : Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
05-23 09:12:19.714 31228 31228 I ScanJob : Using immediateScanJobId from manifest: 208352939
05-23 09:12:19.719 31228 31228 I ScanJob : Using periodicScanJobId from manifest: 208352940
05-23 09:12:19.720 31228 31228 W JobInfo : Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
05-23 09:12:19.720 31228 31228 W JobInfo : Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
05-23 09:12:19.737 31228 31257 E libc    : Access denied finding property "vendor.debug.egl.profiler"
05-23 09:12:19.733 31228 31228 W org.altbeacon.beaconreference: type=1400 audit(0.0:111759): avc: denied { read } for comm=45474C20496E6974 name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=23052 scontext=u:r:untrusted_app:s0:c238,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
05-23 09:12:19.766 31228 31228 D MonitoringActivity: onCreate
05-23 09:12:19.810 31228 31228 D OpenGLRenderer: Skia GL Pipeline
05-23 09:12:19.904 31228 31228 I ScanJob : Using immediateScanJobId from manifest: 208352939
05-23 09:12:19.908 31228 31228 I ScanJob : Using periodicScanJobId from manifest: 208352940
05-23 09:12:19.908 31228 31228 W JobInfo : Requested interval +5m0s0ms for job 208352940 is too small; raising to +15m0s0ms
05-23 09:12:19.908 31228 31228 W JobInfo : Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
05-23 09:12:19.976 31228 31274 I Adreno  : QUALCOMM build                   : 97eb455, I987d6683f9
05-23 09:12:19.976 31228 31274 I Adreno  : Build Date                       : 11/29/18
05-23 09:12:19.976 31228 31274 I Adreno  : OpenGL ES Shader Compiler Version: EV031.25.03.02
05-23 09:12:19.976 31228 31274 I Adreno  : Local Branch                     : 
05-23 09:12:19.976 31228 31274 I Adreno  : Remote Branch                    : 
05-23 09:12:19.976 31228 31274 I Adreno  : Remote Branch                    : 
05-23 09:12:19.976 31228 31274 I Adreno  : Reconstruct Branch               : 
05-23 09:12:19.976 31228 31274 I Adreno  : Build Config                     : S P 4.0.10 AArch64
05-23 09:12:19.979 31228 31274 E libc    : Access denied finding property "ro.vendor.graphics.memory"
05-23 09:12:19.973 31228 31228 W RenderThread: type=1400 audit(0.0:111760): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=23052 scontext=u:r:untrusted_app:s0:c238,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
05-23 09:12:19.980 31228 31274 I Adreno  : PFP: 0x016ee176, ME: 0x00000000
05-23 09:12:19.983 31228 31274 I ConfigStore: android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasWideColorDisplay retrieved: 1
05-23 09:12:19.984 31228 31274 I ConfigStore: android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasHDRDisplay retrieved: 1
05-23 09:12:19.984 31228 31274 I OpenGLRenderer: Initialized EGL, version 1.4
05-23 09:12:19.985 31228 31274 D OpenGLRenderer: Swap behavior 2
05-23 09:12:20.015 31228 31228 I AssistStructure: Flattened final assist data: 2500 bytes, containing 1 windows, 7 views
05-23 09:12:20.028 31228 31274 E libc    : Access denied finding property "vendor.gralloc.disable_ahardware_buffer"
05-23 09:12:20.023 31228 31228 W RenderThread: type=1400 audit(0.0:111761): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=23052 scontext=u:r:untrusted_app:s0:c238,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
05-23 09:12:20.100 31228 31228 I CycledLeScanner: Using Android O scanner
[... cut ...]
05-23 09:25:11.580 31228 31228 I ScanJob : Scan job running for 300000 millis
05-23 09:25:11.581 31228   616 D BluetoothAdapter: isLeEnabled(): ON
05-23 09:25:11.583 31228 31245 D BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=14 mScannerId=0
05-23 09:25:18.203 31228   616 D BluetoothAdapter: isLeEnabled(): ON
05-23 09:25:18.207 31228   616 D BluetoothAdapter: isLeEnabled(): ON
05-23 09:25:18.211 31228 31245 D BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=14 mScannerId=0
```

mohand1993, 2019-05-23:
Thanks for your quick response.
I re-checked that, and yes you're right the scanning did start again after ~11 min.
Anyway I think that the we can get over the android 8+ limitation by  starting passive scan.
Please have a look at https://github.com/AltBeacon/android-beacon-library/pull/882

davidgyoung, 2019-05-23:
Ok, I'm going to close this issue as the region exit does happen as designed (although admittedly not as quickly ad you'd like!)  Please see #869 for a discussion of the challenges of getting rapid region exit events on Android 8+.  Unfortunately, I think it is not possible.

closed 2019-05-23 by davidgyoung