AltBeacon/android-beacon-library #883: CycledLeScannerThread leaks on Android O

JerryLocke, 2019-05-26:
The version number I am using is 2.16.1.

Due to the need to manually start/stop beacon scanning frequently, and in order to speed up the scanning speed, we call ```BeaconManager.bind(this)``` in ```Application.onCreate()``` to start ```BeaconService``` by default.

I noticed that when I first created the ```BeaconManager``` singleton, the library would call ```setScheduledScanJobsEnabledDefault()```, which set ```mScheduledScanJobsEnabled``` to true on Android O. This will start the ```ScanJob`` to scan beacons, and this is what I don't need. I think this value should be false in most scenarios to save power. Unfortunately, there is no relevant setting to turn off this feature.

Under the above conditions, the Beacon scan will be triggered continuously. But it seems that the library does not handle this situation very well, resulting in the creation of a large number of scan threads, and did not get closed in time.
[image]
This leads to the creation of a large number of file descriptor, which ultimately reaches the limit of a maximum of 1024 file descriptor for a single process in the Android system. And throwing ```java.lang.RuntimeException: Could not read input channel file descriptors from parcel.``` exception, causing the app to crash.

I am eager to get your help to improve this library and fix above problems. Thank you!

davidgyoung, 2019-05-26:
A few points:

*  `setScheduledScanJobsEnabledDefault` simply checks if the Android OS version is 8+ (requiring job scheduler to do work in the background vs long-running services) to set the default setting.  You can always override its default by calling `beaconManager.setScheduledScanJobsEnabled(false)` if you don't want to use scan jobs.

* I'm not sure exactly what you mean by the need to  "start/stop beacon scanning frequently". The library certainly supports stopping ranging and monitoring (and even unbinding) and then later on binding and starting ranging and monitoring.  I have many apps that do this.  But I can certainly imagine that if you unbind, then rebind really fast -- say a few milliseconds later -- this would cause problems.  Starting up a JobService on Android is not instantaneous, nor is shutting it down and cleaning it up.

* I'm certainly open to any simple fixes that would make starting and stopping scanning quickly more robust.  But this also sounds like a pretty unusual case that might be better avoided by not starting and stopping scanning so quickly.  It reminds me of the old joke:  
[image]

JerryLocke, 2019-05-27:
Good communication is the beginning of everything. I may not have described the problem clearly.

The point is not I need to start and stop the scanning frequently, nor the ```setScheduledScanJobsEnabledDefault``` can turn off the default behavior -- I certainly know this.
I emphasize these above because they are contexts, I think these can help you find out where the problem lies.

As the title says, the focus of the problem is on thread leaks. I found that under the premise of turn on these feature by default and use ```BackgroundPowerSaver```, entering and exiting activities will cause ```CycledLeScannerThread``` to be continuously created, and did not get closed in time. This will eventually lead to the problem I mentioned above.  In addition, under the premise that I turned off the auto-start feature on Android 8+, manually start scanning, after a long period of time, ANR will also happen.

I created a demo project and successfully reproduced the above questions. If you need it, I can send it to you by email.

davidgyoung, 2019-05-27:
Can you reproduce this with the [reference application](https://github.com/AltBeacon/android-beacon-library-reference)?  Or is there a minimal change you can make to that reference application to trigger the issue?

JerryLocke, 2019-05-28:
I just made the following changes to the reference application:
- Added ```beaconManager.bind(this)``` in ```BeaconReferenceApplication```
- Added ```ThreadLeakActivity``` with only one close button, click ```Thread Leak``` button in the MonitoringActivity can open it, and click ```Close``` button can close it.

This problem can be reproduced by the following steps:
1. Click ```Disable monitoring``` button first
1. Open and close ```ThreadLeakActivity``` repeatedly and observe the thread information in the debugger panel.

Related changes please follow this link: https://github.com/JerryLocke/android-beacon-library-reference/commit/e51921a1a341488fc0522b4712feba3322a7fab3

davidgyoung, 2019-05-29:
@JerryLocke I have reproduced this issue -- it's actually a bit easier to reproduce than you say, and can be reproduced with an unmodified reference application on Android 8+:

1. Launch the reference app
2. Tap the stop monitoring button
3. Hit the home button to put the app to the background
4. Use the task switcher to switch back to the app
5. Notice in the debugger that a new CycledLeScanner thread is started
6. Repeat steps 3-5 to see multiple threads get created

The problem is triggered by the use of `BackgroundPowerSaver` when there are no consumers bound to the `BeaconManager`.  Each time the app switches from the background to the foreground, the `BackgroundPoserSaver` alerts the scan periods for foreground or background mode, then if the JobScheduler is being used for scans (as it is by default on Android 8+) it will reschedule scans to apply the settings changes.  This starts up a new ScanJob when the app is in the foreground, which is what causes the thread to be created.

The bug is inside BeaconManager.  It should not try to apply the settings changes to the scheduled jobs if no consumers are bound.  Doing so incorrectly causes the thread leak.  This is fixed in #885

closed 2019-05-29 by davidgyoung

JerryLocke, 2019-05-30:
The version 2.16.2 did fix the problem with the unmodified reference application, but it still exists in the app I modified in my previous comment.
I mentioned above that I bound the ```BeaconManager``` when the app was created. So even if there is a consumer, switching the app from the background to the foreground will still cause the thread leak.

JerryLocke, 2019-05-30:
Also, does this cause changing scan period not work when no consumers are bound?

davidgyoung, 2019-05-30:
@JerryLocke, I looked at your case again and found it to be slightly different.  I have a fix here: #888 Please read that for an explanation.  

To answer your other question, the scan periods should not matter if no consumers are bound because  no data will be returned anyway.

JerryLocke, 2019-05-31:
I just tested that PR and the problem has been completely fixed.

Thank you for fixing this problem.