AltBeacon/android-beacon-library #884: Support for Custom Protocol Format

android-bash, 2019-05-28:
### Expected behavior
It is an enquiry and not a bug/issue. 

I went through https://github.com/AltBeacon/spec and found out support for Altbeacon protocol. Similarly if we use our own custom protocol how can we use this library to support it?
What are the classes that i will have to implement/extend for doing so? 

Can you please give me more info on having a custom protocol and its usage via this library. It would be really helpful. Thanks a lot for this library. 

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-05-28:
For help in using this library, please open a question on StackOverflow.com so that we may leave this forum reserved for bug reports and feature requests.

You may want to review the documentation for the BeaconParser class here: https://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconParser.html#setBeaconLayout-java.lang.String-

If you need additional help, please open a question on StackOverflow.com

closed 2019-05-28 by davidgyoung

android-bash, 2019-05-28:
Sure Thank you. I will do it from next time. But one final question i would like to know here is how the library differentiates between whether Beacon emits iBeacon protocol or eddystone or any other custom one?

android-bash, 2019-06-03:
https://stackoverflow.com/questions/56423694/difference-between-runningaveragerssifilter-armarssifilter-kalman-filter-in-al