AltBeacon/android-beacon-library #889: Attempt to invoke virtual method 'org.altbeacon.beacon.service.MonitoringStatus org.altbeacon.beacon.service.ScanState.getMonitoringStatus()' on a null object reference

PongPloy2016, 2019-06-04:
I use android device HUAWEI Y9 2019  and ran  application   when I call  to connect Beacon  

and wait show error  logcat 

[image]

Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'org.altbeacon.beacon.service.MonitoringStatus org.altbeacon.beacon.service.ScanState.getMonitoringStatus()' on a null object reference
       at org.altbeacon.beacon.service.ScanJob.startPassiveScanIfNeeded(ScanJob.java:139)
       at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:171)
       at android.app.job.JobService$1.onStopJob(JobService.java:76)
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:117)
       at android.os.Handler.dispatchMessage(Handler.java:109)
       at android.os.Looper.loop(Looper.java:166)
       at android.app.ActivityThread.main(ActivityThread.java:7555)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:469)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:963)

jmvines, 2019-06-04:
Same issue here

PongPloy2016, 2019-06-04:
> Same issue here

 Wow  thank you to edit

PongPloy2016, 2019-06-05:
@Ok thank you  @davidgyoung  I ad to edit code   and  I  will try the test again   

Thank you for Answer to help me 

[image]

closed 2019-06-05 by PongPloy2016

davidgyoung, 2019-06-05:
@PongPloy2016, can you please try this beta release that fixes this problem and let me know if you no longer see the crash:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.3-beta1

PR #891 (2019-07-09): Fix IllegalArgumentException from ending user switch watching
Attempt to fix #889