AltBeacon/android-beacon-library #895: Issues with upgrade to gradle 5.x

acutetech, 2019-06-20:
While upgrading a project to "latest" SDK, gradle I got this error:
"Got error: Could not find method leftShift() ..."

Turns out this is fixed by replacing "<<" twice in build.gradle and introducing a doLast{} construction - e.g.

task release(dependsOn: 'distribution')  {
    doLast {
        println('Doing release build')
    }
}

See https://docs.gradle.org/current/userguide/upgrading_version_4.html

I also received a warning about Build Tools version 28.0.0 being out of date: "To suppress this warning, remove "buildToolsVersion '28.0.0'" - which I did.

There was also this warning, which I did not track down and fix:
API 'variant.getJavaCompile()' is obsolete and has been replaced with 'variant.getJavaCompileProvider()'.
 
Also this after attempting upgrade to AndroidX in a project that incorporates the library:
"ERROR: Cannot change strategy of configuration ':android-beacon-library-uuid128:doclava' after it has been resolved."

I did not understand this, but commented out the doclava code.

davidgyoung, 2019-06-29:
Thanks for the reports.  I have been trying to track down these same issues today, but I'm really not a gradle export.  I think I can resolve the other issues you described, but I'm also stuck on the doclava problem.  Given the need to maintain the library, removing the ability to generate JavaDocs (also mentioned in #874) is not a very good solution.

davidgyoung, 2019-06-29:
Please see #900 with my attempt to fix this.  I'd like to fix the remaining warning before merging.  Any help on that is appreciated.