AltBeacon/android-beacon-library #897: App stops scans beacons after several hours in background

turbo-lab, 2019-06-28:
### Expected behavior
It should scan bluetooth every backgroundBetweenScanPeriod

### Actual behavior
It tries to start scan but something goes wrong. I got such logcat message
`BtGatt.GattService: App 'org.turbo.beaconmqtt.debug' is scanning too much clientif`
and app goes sleep to next scan period

### Steps to reproduce this behavior
Start app and wait some time

### Mobile device model and OS version
Xiaomi Note 5, Android 9

### Android Beacon Library version
2.16.1, 2.16.2

Below I placed logcat peace that I getting in the cycle:
`2019-06-28 15:07:02.397 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothAdapter: isLeEnabled(): ON
2019-06-28 15:07:02.397 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothLeScanner: could not find callback wrapper
2019-06-28 15:07:02.398 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: starting a new bluetooth le scan
2019-06-28 15:07:02.400 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
2019-06-28 15:07:02.400 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Using an empty scan filter since this is 8.1+ on Non-Samsung
2019-06-28 15:07:02.401 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Waiting to stop scan cycle for another 2000 milliseconds
2019-06-28 15:07:02.402 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothAdapter: isLeEnabled(): ON
2019-06-28 15:07:02.405 2952-12030/? D/BtGatt.GattService: registerScanner() - UUID=ca3d3c9d-0dcc-473c-8383-85dd0f66e2cf
2019-06-28 15:07:02.406 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{92f75b7: android.os.BinderProxy@417283}
2019-06-28 15:07:02.406 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Scan started
2019-06-28 15:07:02.406 2952-12030/? E/BtGatt.GattService: App 'org.turbo.beaconmqtt.debug' is scanning too much clientif
2019-06-28 15:07:02.407 7768-10673/org.turbo.beaconmqtt.debug D/BluetoothLeScanner: onScannerRegistered() - status=6 scannerId=-2 mScannerId=0
2019-06-28 15:07:02.413 3898-4349/? W/PushService: 2019-06-28 15:07:02,413 - [WARN::PushService] - [Thread:2326] [Slim] RCV ping id=0
2019-06-28 15:07:02.416 3898-4218/? W/PushService: 2019-06-28 15:07:02,416 - [WARN::PushService] - [Thread:129] TinyData TinyDataCacheProcessor.pingFollowUpAction ts:1561709222416
2019-06-28 15:07:02.424 3898-4218/? W/PushService: 2019-06-28 15:07:02,423 - [WARN::PushService] - [Thread:129] TinyData TinyDataCacheProcessor.pingFollowUpAction !canUpload(uploader) ts:1561709222423
2019-06-28 15:07:02.875 713-763/? E/ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
2019-06-28 15:07:03.408 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Waiting to stop scan cycle for another 992 milliseconds
2019-06-28 15:07:03.418 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{92f75b7: android.os.BinderProxy@417283}
2019-06-28 15:07:04.412 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Done with scan cycle
2019-06-28 15:07:04.412 7768-7768/org.turbo.beaconmqtt.debug D/ScanHelper: Beacon simulator not enabled
2019-06-28 15:07:04.416 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: stopping bluetooth le scan
2019-06-28 15:07:04.416 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Stopping scan
2019-06-28 15:07:04.424 7768-7809/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
2019-06-28 15:07:04.425 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Normalizing between scan period from 9000 to 8966
2019-06-28 15:07:04.425 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: This is Android L. Preparing to do a filtered scan for the background.
2019-06-28 15:07:04.426 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothAdapter: isLeEnabled(): ON
2019-06-28 15:07:04.426 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothLeScanner: could not find callback wrapper
2019-06-28 15:07:04.426 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: starting filtered scan in SCAN_MODE_LOW_POWER
2019-06-28 15:07:04.427 7768-7768/org.turbo.beaconmqtt.debug D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceSolicitationUuid=null, mServiceSolicitationUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=76, mManufacturerData=[2, 21], mManufacturerDataMask=[-1, -1]]
2019-06-28 15:07:04.429 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothAdapter: isLeEnabled(): ON
2019-06-28 15:07:04.432 2952-4067/? D/BtGatt.GattService: registerScanner() - UUID=a684b082-2f1d-4bc4-84d7-8ce6f1f91e68
2019-06-28 15:07:04.433 2952-4067/? E/BtGatt.GattService: App 'org.turbo.beaconmqtt.debug' is scanning too much clientif
2019-06-28 15:07:04.434 7768-10673/org.turbo.beaconmqtt.debug D/BluetoothLeScanner: onScannerRegistered() - status=6 scannerId=-2 mScannerId=0
2019-06-28 15:07:04.435 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 8966 milliseconds
2019-06-28 15:07:04.444 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{92f75b7: android.os.BinderProxy@417283}
2019-06-28 15:07:05.446 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 7946 milliseconds
2019-06-28 15:07:06.448 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 6944 milliseconds
2019-06-28 15:07:07.450 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 5941 milliseconds
2019-06-28 15:07:08.452 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 4940 milliseconds
2019-06-28 15:07:09.455 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 3936 milliseconds
2019-06-28 15:07:10.458 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 2934 milliseconds
2019-06-28 15:07:11.460 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 1931 milliseconds
2019-06-28 15:07:12.369 3898-4218/? W/PushService: 2019-06-28 15:07:12,368 - [WARN::PushService] - [Thread:129] JOB: check the ping-pong.1561709222363
2019-06-28 15:07:12.462 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 929 milliseconds
2019-06-28 15:07:13.395 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: starting a new scan cycle
2019-06-28 15:07:13.396 7768-7809/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
2019-06-28 15:07:13.400 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothAdapter: isLeEnabled(): ON
2019-06-28 15:07:13.400 7768-7809/org.turbo.beaconmqtt.debug D/BluetoothLeScanner: could not find callback wrapper
2019-06-28 15:07:13.402 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScanner: starting a new bluetooth le scan
2019-06-28 15:07:13.403 7768-7768/org.turbo.beaconmqtt.debug D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
`

turbo-lab, 2019-06-28:
The issue is appears with that parameters:
```
beaconManager.setEnableScheduledScanJobs(false);
...
beaconManager.setBackgroundBetweenScanPeriod(9000);
beaconManager.setBackgroundScanPeriod(2000);
beaconManager.setRegionExitPeriod(3000);
```
I think between scan period is so short. I'll try to increase it.

davidgyoung, 2019-06-28:
The tag on that log line `BtGatt.GattService: App 'org.turbo.beaconmqtt.debug' is scanning too much clientif`  indicates it comes from src/com/android/bluetooth/gatt/GattService.java.

However, in the AOSP code repositories no such line exists.  This suggests that the code is a proprietary closed-source modification made by Xiaomi.  Unfortunately, that means we have to reverse engineer how it works.

turbo-lab, 2019-07-01:
Ok, it looks like platform issue.

closed 2019-07-01 by turbo-lab