AltBeacon/android-beacon-library #898: Support for Androidx

nandinirao, 2019-06-28:
### Expected behavior
We are planning to switch our existing project which uses altbeacon to using Androidx. When is it going to be supported with altbeacon instead of support library ?

### Android Beacon Library version

2.16.2

davidgyoung, 2019-06-29:
@nandinirao, can you please help me understand your use case?  If using a binary distribution, then none of the android support classes are supposed to be exposed (please share specifics if you see otherwise).  For binary distributions therefore, it should not matter when the library's internal build migrates to androidx.

davidgyoung, 2019-06-29:
The official reference app is converted to Androidx here:

https://github.com/AltBeacon/android-beacon-library-reference/pull/56

paolorotolo, 2019-06-29:
@davidgyoung If you switch to Android X, all your Maven dependencies need to depend to Android X artifacts. If not, **jetifier** compatibility mode is enabled and all the artifacts using support libraries are refactored at compile time. This of course slows build time of all the apps that are using ```android-beacon-library```.