AltBeacon/android-beacon-library #902: Bluetooth Medic on EMUI 9

Del-S, 2019-07-01:
Since EMUI 9 there is a dialog to enable/disable bluetooth when it is triggered in code.

### Expected behavior

Bluetooth Medic restarts bluetooth.

### Actual behavior

Bluetooth disable dialog appears.

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-07-01:
@Del-S How did you verify this behavior?

It sounds like you are suggesting that standard Android APIs to power off/on bluetooth have a non-standard implementation on EMUI 9 devices that pops a dialog.  Since I don't have a EMUI 9 device for testing, I'd appreciate any further characterization you can do to find an appropriate behavioral adjustment in the library.  

Worst case, I suppose the library could disable this feature on EMUI 9 devices if it is not possible to restart bluetooth automatically.

Del-S, 2019-07-02:
@davidgyoung I created a quick demo case in our app.

After the application is started following code is triggered.

```java
new Handler().postDelayed(() -> {
    Intent intent = new Intent("onScanFailed");
    intent.putExtra("errorCode", 2);
    LocalBroadcastManager.getInstance(context).sendBroadcast(intent);
}, 1000);
```

Result of this can be seen in the [video](https://drive.google.com/file/d/15pcLNEP3cSa1cNhUQgUVefUlHBBSWG0d/view?usp=sharing). The video shows that bluetooth is enabled. After application is started **BluetoothMedic** receives _onScanFailed_ broadcast and acts accordingly. System displays the dialog to disable bluetooth, if user tap to accept it, bluetooth is disabled but it will never try to enable it again.

Also **BluetoothMedic** disables bluetooth even if there are any active connections (hands-free, playing music over bluetooth) and that is unacceptable for us. These two reasons are why we will be moving away from medic implementation and prompt the user using notification that they need to restart Bluetooth on their own.

We are scanning every minute using foreground service. The plan is to listen for _onScanFailed_ and _onStartFailed_, check if the scan actually failed, no updates were received in  **MonitorNotifier** or **RangeNotifier**. If the scan actually failed we increase the fail count, when the scaning fails more than 5 times in a row we will prompt the user about bluetooth error.

davidgyoung, 2019-07-04:
Thank you for the detail and the video, @Del-S.  Based on your info, I agree that the Bluetooth Medic's feature to cycle power is worse than useless on EMUI 9.   I am not sure the proper way to handle this.  A few thoughts:

1. Add documentation to the method for `medic.enablePowerCycleOnFailures(context);` warning this feature is known to fail on EMUI 9 devices and present the user with a visible popup.

2. In addition to the above, add code to detect a EMUI 9 device and log a warning for the above if the method is called.

3. In addition to the above, add code to refuse to power cycle if a EMUI 9 device is detected.

I'm not sure the best approach.  Option 3 might be most helpful to folks who target a wide range of Android devices but are unaware of this problem on EMUI 9.  But in the event that a different or subsequent release of EMUI no longer engages in this custom behavior, implementing option 3 could make things worse.

I am also a bit reluctant to keep adding platform-specific behaviors in this library to work around vendors' secret non-standard Android behavioral modifications, as it is impossible to know about all of them and difficult to keep up with new ones that are discovered.  Further, such changes won't  app developers with existing apps who don't upgrade the library version regularly.  Given this, is it worth automatically detecting EMUI 9 and changing the behavior?

Del-S, 2019-07-05:
@davidgyoung Yeah Chinese based Android versions always had their custom stuff to deal with like strict battery optimization. I myself discovered this issue accidently by upgrading one of our test devices.

1. I would add it to the function and also to the top of the class. Also it is good to mention that it is known on EMUI 9 but it might be the case for other systems as well, we just might not know it yet.

2. I would rather add log that informs the developer that Medic was not able to cycle bluetooth. Maybe even log when transition between enabled and disabled state fails (Medic tried to disable bluetooth but after few seconds it is still not disabled). This makes sure it is not platform specfic.

3. I too am reluctant to add platform-specific behaviors to any library. Library is providing features and it's up to the app developer to decide if they want to use them or not. So I would not refuse power cycle but I would make few changes to the Medic to make sure it works.

Taking a quick look at the code. It is using _Handler_ to wait before enable bluetooth is triggered. I would change it to the receiver that listens for `BluetoothAdapter.ACTION_STATE_CHANGED` and then add logic to enable bluetooth if the medic triggered bluetooth cycle. On EMUI devices it would show up 2 dialogs right after each other (which is annoying) but at least it would work.

I would also add an option to not power cycle if there are any active bluetooth connections. Hence the case with hands-free or playing musing over bluetooth. App developer would decide if they want to power cycle even if there are active connections or not.