AltBeacon/android-beacon-library #903: Why rssi looks like period?

Sunnystaste, 2019-07-02:
[image]

This picture is a rssi picture of an iBeacon. iBeacon broadcast frequency is 10 Hz. The receiving period of phone is 110ms. The picture looks like period and the fluctuation of the signal is about 15db. It is a disaster to indoor location. Why the rssi signal is period? It is the problem of program or the problem of iBeacon?

fkruta, 2019-07-02:
welcome to the joys of BLE indoor location. The fluctuation are normal due to most likely to MultiPath, now in your case they are significantly big. You might have a damage circuit either on the phone or the beacon.

davidgyoung, 2019-07-03:
IMO that pattern looks like the delta RSSI from scanning on different BLE advertising channels.  See my answer here:  https://stackoverflow.com/questions/49634513/is-there-an-explanation-for-the-regular-oscillation-experienced-in-bluetooth-rss

Please post followup questions on StackOverflow.com as that is the appropriate forum for questions like this.  This forum is reserved for bug reports and feature requests.

closed 2019-07-03 by davidgyoung