AltBeacon/android-beacon-library #904: NPE in ScanJob restartScanning()

franvis, 2019-07-02:
### Actual behavior

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.Set p2.a.a.m.e.b()' on a null object reference
       at org.altbeacon.beacon.service.ScanJob.restartScanning + 256(ScanJob.java:256)
       at org.altbeacon.beacon.service.ScanJob.startScanning + 288(ScanJob.java:288)
       at org.altbeacon.beacon.service.ScanJob.access$500 + 41(ScanJob.java:41)
       at org.altbeacon.beacon.service.ScanJob$1.run + 96(ScanJob.java:96)
       at java.lang.Thread.run + 764(Thread.java:764)

###  Galaxy A5(2017), Android 8.0.0

### Android Beacon Library version: 2.16.3-beta1

davidgyoung, 2019-07-03:
Looks like this is caused by `mScanHelper.getMonitoringStatus()` returning null in the following code

```
if (mScanHelper.getRangedRegionState().size() > 0 || mScanHelper.getMonitoringStatus().regions().size() > 0) {
                if (mScanHelper.getCycledScanner() != null) {
                    mScanHelper.getCycledScanner().start();
                }
                return true;
            }
```

davidgyoung, 2019-07-03:
Fixed in #896 via commit d7b5d2d446936baae0491a38774d0549d248ee7e

closed 2019-07-03 by davidgyoung