AltBeacon/android-beacon-library #907: Saving logs into a file

TayfunCesur, 2019-07-06:
Hi David,

I have made some researches at altbeacon.github.io and also github issues but unfortunately I couldn't find any api info about saving logs. 

I already use BeaconManager.setDebug(true)
But how can I save that logs into a file ?
Is there any api to do this easily or should I have to do myself?
Or should I use adb logcat > someFile.txt

Thanks

davidgyoung, 2019-07-06:
Yes, I typically use adb logcat > someFile.txt.
You can also tail that file in another console window to monitor it while
the log is writing.

TayfunCesur, 2019-07-07:
Thank you David. I also searched the Android Central Log System. I can share what I learned for those who are curious about this.

Android Central Log System keeps the logs until buffer size reaches. The buffer size can be change under developer options up to 16 MB. Mine is 256kb and I can get about last 5 hours log.

Thank you.

closed 2019-07-07 by TayfunCesur

prasad456, 2019-07-18:
how to stop logging i want to aviod scanState.save()