AltBeacon/android-beacon-library #908: AltBeacon S8 Android 9

RobsonTeixeira, 2019-07-11:
Using android-beacon-library-reference.
Using foreground scan (uncommented the code).
Using my beacon filter.

### Expected behavior
App runing ok, after locking screen keeps running.

### Actual behavior
After lock screen, it stops right away.

### Mobile device model and OS version
Galaxy S8 on Android 9.
Works fine on A5 2017 on Android 8.

### Android Beacon Library version
Newer (Marked as 2+)

After some research I found out that on Android 9, on some Samsung devices, you need to create a valid ScanFilter before scanning, could that be the problem?

davidgyoung, 2019-07-11:
Please check your specific library version number.  Samsung 8.1+ requires a non-empty BLE scan filter for background detections.  These were added as of 2.15.3.

RobsonTeixeira, 2019-07-11:
Using version 2.16.2

Here the log from the S8, first few lines when still working, then the phone is locked and stops right away. On the A5 it keeps going normally.

`2019-07-11 12:25:14.044 11471-11471/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
2019-07-11 12:25:14.044 11471-11471/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
2019-07-11 12:25:14.044 11471-12043/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1eff06000109200230625bea2652067af15cb4adf79a2b1e5e6e9659b3431400000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
2019-07-11 12:25:14.044 11471-12045/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1eff06000109200079b0a1f94ea110595e75f6853116a8ab1ca6fe7aae3a3c00000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
2019-07-11 12:25:14.044 11471-12043/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting 06 07.  The bytes I see are: 1eff06000109200230625bea2652067af15cb4adf79a2b1e5e6e9659b3431400000000000000000000000000000000000000000000000000000000000000
2019-07-11 12:25:14.044 11471-12045/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting 06 07.  The bytes I see are: 1eff06000109200079b0a1f94ea110595e75f6853116a8ab1ca6fe7aae3a3c00000000000000000000000000000000000000000000000000000000000000
2019-07-11 12:25:14.144 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
2019-07-11 12:25:14.144 11471-11471/org.altbeacon.beaconreference D/ScanHelper: Beacon simulator not enabled
2019-07-11 12:25:14.146 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 1680 millisconds.
2019-07-11 12:25:14.146 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
2019-07-11 12:25:14.147 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: restarting a bluetooth le scan
2019-07-11 12:25:14.148 11471-11471/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
2019-07-11 12:25:14.149 11471-11471/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Using a non-empty scan filter since this is Samsung 8.1+
2019-07-11 12:25:14.150 11471-11471/org.altbeacon.beaconreference D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=4c, mManufacturerData=, mManufacturerDataMask=]
2019-07-11 12:25:14.150 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
2019-07-11 12:25:14.151 11471-11911/org.altbeacon.beaconreference D/BluetoothAdapter: STATE_ON
2019-07-11 12:25:14.155 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{3a2f0ba: android.os.BinderProxy@b955b6b}
2019-07-11 12:25:14.155 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Scan started
2019-07-11 12:25:14.155 11471-11471/org.altbeacon.beaconreference E/CycledLeScannerForLollipop: Scan failed: a BLE scan with the same settings is already started by the app
2019-07-11 12:25:14.346 11471-11471/org.altbeacon.beaconreference D/ViewRootImpl@97e4c72[MonitoringActivity]: Surface release. android.view.ViewRootImpl$ViewRootHandler.handleMessage:4802 android.os.Handler.dispatchMessage:106 
2019-07-11 12:25:14.377 11471-11909/org.altbeacon.beaconreference W/libEGL: EGLNativeWindowType 0x75d3a39010 disconnect failed
2019-07-11 12:25:14.377 11471-11909/org.altbeacon.beaconreference D/OpenGLRenderer: eglDestroySurface = 0x75be779f00, 0x75d3a39000
2019-07-11 12:25:14.398 11471-11471/org.altbeacon.beaconreference D/ViewRootImpl@97e4c72[MonitoringActivity]: Relayout returned: old=[0,0][1080,2076] new=[0,0][1080,2076] result=0x5 surface={valid=false 0} changed=false
2019-07-11 12:25:14.403 11471-11471/org.altbeacon.beaconreference D/ViewRootImpl@97e4c72[MonitoringActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2019-07-11 12:25:14.404 11471-11471/org.altbeacon.beaconreference D/InputMethodManager: prepareNavigationBarInfo() DecorView@5607810[MonitoringActivity]
2019-07-11 12:25:14.404 11471-11471/org.altbeacon.beaconreference D/InputMethodManager: getNavigationBarColor() -855310
2019-07-11 12:25:15.155 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 95 milliseconds
2019-07-11 12:25:15.160 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{3a2f0ba: android.os.BinderProxy@b955b6b}
2019-07-11 12:25:15.255 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
2019-07-11 12:25:15.255 11471-11471/org.altbeacon.beaconreference D/ScanHelper: Beacon simulator not enabled
2019-07-11 12:25:15.256 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 4894 millisconds.
2019-07-11 12:25:15.256 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
[... cut ...]
2019-07-11 12:25:17.441 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 60 milliseconds
2019-07-11 12:25:17.459 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{3a2f0ba: android.os.BinderProxy@b955b6b}
2019-07-11 12:25:17.520 11471-11471/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
2019-07-11 12:25:17.520 11471-11471/org.altbeacon.beaconreference D/ScanHelper: Beacon simulator not enabled`

RobsonTeixeira, 2019-07-12:
Problem solved.

Using setHardwareAssistManufacturerCodes() on BeaconParser to explicit set the manufacturer code.

closed 2019-07-12 by RobsonTeixeira

android-bash, 2019-08-21:
Should setHardwareAssistManufacturerCodes() be used in my app as well? I have samsung users.