AltBeacon/android-beacon-library #910: 2.16.3 release

venjirai, 2019-07-14:
I'm using 2.16.2 and I'm seeing many crashes from my users in crashlytics using Android 8 and 9.

null object reference
org.altbeacon.beacon.service.ScanState.getMonitoringStatus (ScanState.java:99)
org.altbeacon.beacon.service.ScanJob.startPassiveScanIfNeeded (ScanJob.java:139)
org.altbeacon.beacon.service.ScanJob.onStopJob (ScanJob.java:171)

null object reference
org.altbeacon.beacon.service.ScanJob.restartScanning (ScanJob.java:231)
org.altbeacon.beacon.service.ScanJob.startScanning (ScanJob.java:259)
org.altbeacon.beacon.service.ScanJob.access$500 (ScanJob.java:40)

null object reference
org.altbeacon.beacon.service.ScanJob.stopScanning (ScanJob.java:180)
org.altbeacon.beacon.service.ScanJob.onStopJob (ScanJob.java:170)
android.app.job.JobService$1.onStopJob (JobService.java:67)

I have read that 2.16.3 will fix those issues but I'm unable to use the pre-release beta4 version as gradle says it cannot be found.
So my question is, when will 2.16.3 be available?

FongMi, 2019-07-18:
import same as 2.16.3-beta3, just follow https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.3-beta3

shawnvega, 2019-07-24:
Do we know when the 2.16.3 release will happen? In general, we try not to use beta code in our production app.

venjirai, 2019-07-26:
Any update?
Why can't the beta version also just be included without manually adding the library files?

davidgyoung, 2019-07-26:
@shawnvega ad @venijrai, we really need to have sufficient testing on this beta before we make a formal release, otherwise unsuspecting app developers may end up with crashes due to insufficient testing.

What makes this complicated is that the bug fixes in 2.16.3 beta builds all come out of crash logs without specific steps to reproduce.  So we end up with a chicken and egg problem:

1. People get crash logs.
2. This project implements bug fixes to resolve the crash logs, and issues a 2.15.3 beta release.
3. This project waits for sufficient testing feedback that the build is stable and that the bugs are resolved.  Because these specific bugs can only be reproduced in production through crash logs we must have developers willing bundle the beta release with their app and send it to the public to verify.
4. Developers like @shawnvega do not do #3 because they want to wait for a non-beta release.
5. Repeat from step 3.

As you can see, we are in an endless loop!  The release is delayed until we can get sufficient testing.  Understand that this library is an open source project.  We don't charge anybody for its use.  In return, we really need assistance from the community that uses it in testing these bug fixes, and that means being willing to deploy releases with beta software and report back results.

**In order to get this release out, I ask that you please:**

(1) Follow the instructions here to use the latest beta release with your app: https://github.com/AltBeacon/android-beacon-library/releases/tag/2.16.3-beta4

(2) Test on your own devices and verify proper behavior.

(3) Issue a release to the public with the beta software above, and watch closely for crash logs for the next day or so.

(4) Report back to this issue your testing results.

If you are unwilling to do the above, you are welcome to continue using the latest non-bet release 2.16.2.   Only once we have sufficient testing by affected users will these changes be put into a formal release.  So far I have feedback from exactly one user, and that is not sufficient.

venjirai, 2019-07-26:
I like to test it for my production build but what's the reason we have to download the library manually?
It's labeled beta, so people are fully aware the build can be unstable

davidgyoung, 2019-07-26:
@venjirai, it is quite easy.  Did you try the instructions?

venjirai, 2019-07-26:
It's easy but it's another thing to remember to remove when updating to a stable version. Also it has to be added to version control. I have not seen other libraries going with that approach either, they all just make their beta versions accessible with that one gradle line.

shawnvega, 2019-08-13:
@davidgyoung It would be nice to not have to commit that to version control, and not to have to manually delete the file when it's not in beta anymore.

travistkachuk, 2019-08-14:
@davidgyoung I followed your instructions for adding the Beta for my project,  however it is failing to resolve the dependency. Any way you could host this?

shawnvega, 2019-08-19:
@davidgyoung We will be running the new beta version of your altbeacon library in our production app for a large theme park. We will report back here after it is deployed.

shawnvega, 2019-08-29:
@davidgyoung We've run the new beta build in many production apps across the US and Mexico. This new beta is a lot more stable than the previous versions. It does have this crash, but it is not as frequent as the previous versions.  
Fatal Exception: java.util.ConcurrentModificationException
       at java.util.ArrayList$Itr.next + 860(ArrayList.java:860)
       at org.altbeacon.beacon.service.ScanJob$1.run + 81(ScanJob.java:81)
       at java.lang.Thread.run + 764(Thread.java:764)

venjirai, 2019-09-11:
Still no release? It's been nearly 2 months now.

Alqueraf, 2019-09-16:
We have added the beta version to our ~30k users app last week, so far the crashes have not reappeared. Will update if any issue comes up.

davidgyoung, 2019-09-16:
Thank you @Alqueraf and @shawnvega.  I have also tested successfully in an app in the Play store.  I will roll a release this week.

davidgyoung, 2019-09-18:
Released to JCenter today

https://bintray.com/altbeacon/android/android-beacon-library/2.16.3

closed 2019-09-18 by davidgyoung