AltBeacon/android-beacon-library #911: Hide Foreground Notification when bluetooth is OFF

mgursch, 2019-07-22:
### Expected behavior
If the user disables bluetooth, the foreground notification should disappear.
### Actual behavior
If the user disables bluetooth, the foreground notification does not disappear.

### Steps to reproduce this behavior
Create a foreground scanning service and disable bluetooth

### Mobile device model and OS version
Android

### Android Beacon Library version
2.16.2

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-07-22:
This is a reasonable feature request.  If you would like to work on it, I would welcome a pull request to implement it.  If you would like tips on doing so, feel free to comment here.

mgursch, 2019-07-22:
Thanks for your response. i would say this is a bug report, because there is no reason to say "scanning for beacons" when bluetooth is off. users are not able to understand that.

davidgyoung, 2019-07-22:
Sorry, the library simply does not supply this feature out of the box.  The intended design is for the application programmer to be in control of when the foreground service is running through beaconManager.bind()/unbind() calls.  You can certainly write application code to stop the foreground service in this case, if you wish.

mgursch, 2019-07-23:
thanks a lot :)

paulpv, 2020-01-24:
I was tempted to open a new Issue in this area, but I'll try piggybacking on this one first.
I am considering using this library, but an issue in this area is currently blocking me.

I disagree that the foreground notification should disappear if BT is turned off.
In my app's (that currently does not use this library) use case I detect when BT is turned off and then change my foreground notification to say "Waiting for Bluetooth...".
This way my app is not killed by the OS while waiting for BT, such that when BT is eventually turned back on my app will be running and can immediately onReceive BT turning on and resume scanning.

This library seems to not make this possible.
I think the core problem is that `enableForegroundServiceScanning` that sets the notification must be called when there are no consumers (ie, only at startup?).
```
    public void enableForegroundServiceScanning(Notification notification, int notificationId)
            throws IllegalStateException {
        if (isAnyConsumerBound()) {
            throw new IllegalStateException("May not be called after consumers are already bound.");
        }
        ...
```
This effectively limits the notification to being static text.

I propose making it so that `enableForegroundServiceScanning ` can be called dynamically, **and/or** that the notification is a callback method to build the notification when needed.

I can imagine that attempting to do this might open a can of worms. :/

davidgyoung, 2020-01-24:
What is the general procedure for changing the text of a notification behind a foreground service once it is presented?  I imagine you don't have to stop and restart the foreground service.

paulpv, 2020-01-24:
In a perfect world you would only need to find some simple way to call `BeaconService`'s `startForeground` with the new requestCode and notification.
If you bind to the service, which you do in `RegionBootstrap`, then you could save a reference to the service to allow `startForeground` to be directly called.

I may attempt to expose this in my fork of your repo.

davidgyoung, 2020-01-24:
@paulpv, I think we are making this harder than it needs to be.  Based on this [answer](https://stackoverflow.com/a/20142620/1461050),  It seems you simply need to keep track of the notificationId you use for the original notification.  Then you can update it with a new notification like this:

```
    NotificationManager mNotificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
    mNotificationManager.notify(notificationId, notification);
```

No need to restart the service or do anything complicated.  If the above works no modification to the library is needed.

paulpv, 2020-01-24:
Thanks.
I'll check it out.

paulpv, 2020-01-27:
Affirmative. I tweaked my fork of your android-beacon-library-reference to detect BT off/on and update the notification via `notify` and it works splendid.
Thanks for refreshing my memory of a simpler way!

So, this bug **could** be closed and leave it up to the calling developer to keep track of BT state and hide and re-add the notification as desired.
This wouldn't be 100% desirable because this allows the developer to pull the notification out from under the library which doing so could have side-effects on the code and would also defeat the whole purpose of a `foregroundService` that is intended to keep the app process running in the background.

What I did not see in your library is any built-in reusable class to monitor the BT state.
The only BroadcastReceiver I saw in your library that references `BluetoothAdapter.ACTION_STATE_CHANGED` is `BluetoothCrashResolver`.

Perhaps your library could add a handy type of `BluetoothAdapterStateBroadcastReceiver` utility class?

paulpv, 2020-01-27:
FYI, I pushed my test app fork.
https://github.com/paulpv/android-beacon-library-reference/tree/pebblebee/app/src/main/java/org/altbeacon/beaconreference

davidgyoung, 2020-01-27:
Glad to hear that works, @paulpv.  In terms of adding the extra Bluetooth on/off detection utility, I am trying to walk the proper line between adding just enough to make the library easy to use for novices (the justification for the built-in foreground service) but not adding dunctionality that is not core to beacon detection.

Others have criticized the built-in foreground service as being inappropriate for the library, as it should be the responsibility of the app developer.  I am reluctant to add more features to the foreground service that can be handled by the app developer as-is.  To me, it seems the bluetooth on/off notification change fits this description.

Yes, it does require Bluetooth code outside the library to detect on/off.  But this library imakes no attpy to be a Bluetooth replacement module.  I build lots of apps using BLE GATT services that use this library for beacons detection and transmission and raw Bluetooth APIs for GATT communications.