AltBeacon/android-beacon-library #914: Not receiving advertisements when screen off on older phones?

maxkunes, 2019-08-04:
Using the reference application and viewing adb, I seem to infrequently receive advertisements when the phone screen is off. I've seen this happen on my S6 and S7 but not with the exact same app on my S10.

The only code I have modified is adding a beacon parser with this info. This seems to work fine as when the screen is on I can clearly see advertisments come in.
```
mHardwareAssistManufacturers = new int[] {0x0551};
this.setBeaconLayout("m:0-1=5105,i:0-1,p:1-1");
this.mIdentifier = "sylerobeacon";
```

And this to enable continuous scans. 
```
beaconManager.setEnableScheduledScanJobs(false);
beaconManager.setBackgroundBetweenScanPeriod(0);
beaconManager.setBackgroundScanPeriod(1000);
```

I also don't see anything obviously wrong with any log messages.

davidgyoung, 2019-08-04:
You might try setting the long running scan override flag here:

https://github.com/AltBeacon/android-beacon-library/pull/529

maxkunes, 2019-08-04:
@davidgyoung This seems to be enabled by default in the reference project but the comment seems to make a confusing suggestion.

```
<!-- uncomment the following to enable scanning for over 30 minutes on Android 7+ -->
        <service android:name="org.altbeacon.beacon.service.BeaconService"
            tools:node="replace">
            <meta-data android:name="longScanForcingEnabled" android:value="true"/>
        </service>
```

maxkunes, 2019-08-04:
And to clarify, it doesn't seem to do anything for me here.

maxkunes, 2019-08-04:
And to emphasize, a scan will process exactly every 5 seconds after the screen turns off.

davidgyoung, 2019-08-05:
I certainly cannot explain the every 5 second part.  I have done plenty of testing on Samsung Galaxy X devices with the screen off. (Pre-Android 8) and seen detections every second.

There must be  something  else different in your case and the unusual beacon seems like a lot may be a candidate for some unknown reason.

Can you reproduce this 5s problem if you use iBeacon or AltBeacon?  (You can use a beacon simulator app for iOS of Android to try.)

maxkunes, 2019-08-05:
@davidgyoung Well, I can confirm to you its not the beacons fault as I never actually pair to the beacon and listen only for adverts. Having my S6 and S10+ side by side sleeping, the S10+ receives every advert instantly while the S6 very intermittently. 

I installed an altbeacon simulator on my S10+ and broadcasted so my S6 would pick it up. In the log below you can see a log of every time I receive an advert. (Note I added a extra log inside the library to be sure of this, that is what onBeaconAdvert is). I see this same behavior without edits aswell. 

You can clearly see how frequent I receive adverts and then about half way through the log, the phone screen turns off and we get infrequent adverts (once every 5 seconds).
```
2019-08-04 19:14:43.888 13503-14125/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:44.176 13503-14116/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:44.465 13503-14126/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:44.739 13503-14120/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:45.042 13503-14116/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:45.319 13503-14122/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:45.610 13503-14120/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:45.903 13503-14125/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:46.194 13503-14104/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:46.475 13503-14117/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:46.741 13503-14120/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:47.022 13503-14125/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:47.309 13503-14104/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:47.887 13503-14120/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:48.181 13503-14123/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:49.312 13503-14104/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:49.597 13503-14111/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:54.238 13503-14122/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:54.506 13503-14120/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:14:59.390 13503-14125/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:15:04.264 13503-14116/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:15:04.525 13503-14122/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:15:09.688 13503-14111/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
2019-08-04 19:15:14.249 13503-14117/org.altbeacon.beaconreference V/MyBeaconManager: onBeaconAdvert
```
Here you can see another run with the plain library with setDebug(true). I have filtered for "processing pdu type" in this log. Again I let it run with screen on for a few seconds then turn the screen off. We see the exact same behavior.
```
2019-08-04 19:18:28.396 16340-16963/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1bffffffbeacc29ee4c9bd6144dc80e36afb1d2f8c0700000000bb0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 27
2019-08-04 19:18:28.689 16340-16967/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1bffffffbeacc29ee4c9bd6144dc80e36afb1d2f8c0700000000bb0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 27
2019-08-04 19:18:29.003 16340-16880/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1bffffffbeacc29ee4c9bd6144dc80e36afb1d2f8c0700000000bb0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 27
2019-08-04 19:18:29.268 16340-16959/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1bffffffbeacc29ee4c9bd6144dc80e36afb1d2f8c0700000000bb0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 27
2019-08-04 19:18:29.507 16340-16968/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 02011a1bff75004204010176fc039f6d978afe039f6d9789012000000000001e085b54565d2053616d73756e67204672616d652053657269657320283635 with startIndex: 5, endIndex: 30
2019-08-04 19:18:29.545 16340-16971/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 1bffffffbeacc29ee4c9bd6144dc80e36afb1d2f8c0700000000bb0000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 27
2019-08-04 19:18:29.746 16340-16970/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 02011a1bff75004204012076180700020100000000000000000000000000001e085b54565d2053616d73756e67204672616d652053657269657320283635 with startIndex: 5, endIndex: 30
2019-08-04 19:18:29.765 16340-16972/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 02011a0aff4c0010051318417d22000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 13
[... cut ...]
2019-08-04 19:20:24.960 16340-16340/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 1000 milliseconds
2019-08-04 19:20:24.973 16340-16340/org.altbeacon.beaconreference D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{2b5df78: android.os.BinderProxy@2d2eeb6}
2019-08-04 19:20:24.973 16340-16340/org.altbeacon.beaconreference D/CycledLeScanner: Scan started
```

You can see that the libs code is running, so the app seems to be still running fine. Just not getting BLE adverts.

davidgyoung, 2019-08-05:
Hmm.  This is curious as.it is not consistent with my own tests.  Just to eliminate any differences, do you see the same results with the reference app unmodified except to set the background scan period/between scan period to 1000/0?

maxkunes, 2019-08-05:
@danialgoodwin 

Not sure exactly what you mean.

This is what the current code is:
```
beaconManager.enableForegroundServiceScanning(builder.build(), 456);
beaconManager.setEnableScheduledScanJobs(false);
beaconManager.setBackgroundBetweenScanPeriod(0);
beaconManager.setBackgroundScanPeriod(1000);
```

davidgyoung, 2019-08-05:
Try removing these two lines:

beaconManager.enableForegroundServiceScanning(builder.build(), 456);
beaconManager.setEnableScheduledScanJobs(false);

They should not be necessary prior to Android 8.

And please look for any other lines that differ from the default reference app.  I am just trying to eliminate any differences from what I have repeatedly tested before to find what explains the difference between what you see and what I do.

maxkunes, 2019-08-05:
@davidgyoung Same thing with those lines commented.

maxkunes, 2019-08-06:
Huh @davidgyoung I'm not seeing this happen on my S10+. When the screen is off, I no longer receive advertisements. Works fine on the lock screen though. I'm using a completely stock version of the reference app apart from enabling the foreground service code.

davidgyoung, 2019-08-06:
What is the OS version on the S10+?  What is different from your earlier report where you said you saw scans fine on the S10 with the screen off?

maxkunes, 2019-08-06:
Android 9 Kernel 4.14.78-15672595. The only possible thing different from what I can tell is the phone is currently charging at around 20% battery. I have, however, disabled battery optimizations for the sample app...

maxkunes, 2019-08-06:
Huh, so I think I found out what is happening on the S10+. On my older code that is using standing BLE calls, everything works fine. But it seems the android beacon library is acting as if I tried to run a scan without a ScanFilter attached to the scan. My old code with a scan filter works on this device when slept, but my old code without a ScanFilter doesn't work when the device sleeps. I know on the newer versions of android, a scan filter is necessary when using SCAN_MODE_LOW_LATENCY or at all to get events when the device sleeps. @davidgyoung

davidgyoung, 2019-08-06:
Yes, that could explain it.  Starting with Android 8.1 a scan filter is required to detect with the screen off.  The library detects 8.1+ and adds an empty scan filter which works on non-Samsung devices with 8.1+.

Samsung made a proprietary enhancement affecting 8.1+ requiring thos same scan filter to be non-empty.  This means the hardware assist manufacturer must be correct on Samsung 8.1+ to detect in the background

maxkunes, 2019-08-06:
Alright, I'm guessing my hardware assist manufacturer is incorrect then. 
On my older codebase I set that up as so:
`filter.setManufacturerData(1361, byteArrayOf())`
and on the new codebase using this library I do this:
```
class CustomParser internal constructor() : BeaconParser() {
    init {
        mHardwareAssistManufacturers = intArrayOf(0x0551)
        this.setBeaconLayout("m:0-1=5105,i:0-1,p:1-1")
        this.mIdentifier = "sylerobeacon"
    }
}
```
Any idea what could be wrong?

davidgyoung, 2019-08-06:
Use a beacon layout that does not have index offsets less than 2.  This is bad: m:0-1=5105,i:0-1,p:1-1

The above might work for non filtered scans but it will not work for filtered ones.  I should add a warning message to the library about this.

The offsets prior to index 2 are outside the data range of the manufacturer advert and are probably corrupting the scan filter.    If needed, check the system level log messages after the scan starts in the background.  It shows the raw bytes olf the  filter.

maxkunes, 2019-08-06:
> Use a beacon layout that does not have index offsets less than 2. This is bad: m:0-1=5105,i:0-1,p:1-1
>
> The above might work for non filtered scans but it will not work for filtered ones. I should [...]

That seems to fix it on the S10+. I'm extremely confused about why I seemed to have saw this working earlier on the S10+. I may have made a mistake regarding that. I will say, unfortunately, that the issue still persists on the S6 and S7. Could this possibly be caused by Doze mode? As I'm pretty sure those only effect Android 6 and 7?

android-bash, 2019-08-20:
@davidgyoung @maxkunes I am facing this issue on my Google Pixel which was recently upgraded to Android 9. I am not getting beacon scans when screen goes off for few minutes. How do i fix this ?

davidgyoung, 2019-08-20:
@android-bash, please help us keep this issue on target (it is specific to Samsung phones) by posting questions about Pixel phones elsewhere.  I'd suggest that Stackoverflow.com is the best place to start your question about your code.  If we find there is a bug in the library we can open a new issue here.  I have deleted your comment on this issue to keep this on topic, but feel free to copy it and past it as a new SO question and I will try to help there.

android-bash, 2019-08-21:
> @android-bash, please help us keep this issue on target (it is specific to Samsung phones) by posting questions about Pixel phones elsewhere. I'd suggest that Stackoverflow.com is the best place to [...]

Sure thing. I understand. I have opened a question in Stack overflow. Please look in to it. 
https://stackoverflow.com/questions/57589378/how-changes-in-scan-periods-will-affect-continuous-scanning-background-and-froe

paulpv, 2020-01-27:
Close as [possibly] fixed by #941 ?

davidgyoung, 2020-01-27:
@maxkunes can you see if the latest library release with #941 fixes this on older Samsung phones in your tests?

maxkunes, 2020-02-01:
> @maxkunes can you see if the latest library release with #941 fixes this on older Samsung phones in your tests?

I can't do that as I don't have access to the hardware/software anymore as the job came to an end. Sorry!

davidgyoung, 2020-02-01:
Thanks @maxkunes   I believe that change should fix these issues.  Closing until we have a report otherwise.

closed 2020-02-01 by davidgyoung