AltBeacon/android-beacon-library #915: Monitoring: Kontakt.IO beacon didExitRegion when screen gets locked (Samsung A8 / Android 9)

fims, 2019-08-09:
### Expected behavior
Monitoring a specific Kontakt.io beacon while screen gets turned off, didExitRegion should not be called. 

### Actual behavior
Kontakt.io beacons are lost if the screen gets turned off. This only happens with Kontakt.io beacons. If the monitoring region is set to iBeacons, the region is still active if the screen gets locket.

Also when checking with ranging, iBeacons and Eddystone beacons are still visible when screen is locket. Only Kontakt.io beacons are missing.

### Steps to reproduce this behavior
Reference Application with Kontakt.io beacon layout and the following region:
```
private static byte[] test = "ftep48".getBytes();
private static final Identifier identifier = Identifier.fromBytes(test, 0, test.length, false);
private static final Region monitoringRegion = new Region("KontaktIoRegion", identifier, null, null);
```
### Mobile device model and OS version
 Works as expected on:
 - Essential Phone (Android Q Beta 5/6)
 - Samsung A8 (Android 8)

Doesn't work on Samsung A8 with Android 9. 

### Android Beacon Library version
org.altbeacon:android-beacon-library:2.16.2

### Logcat
[Logcat.txt](https://github.com/AltBeacon/android-beacon-library/files/3486122/Logcat.txt)

davidgyoung, 2019-08-09:
I suspect the issue may be that your beacon has a non-standard manufacturer code.  If so, you will need to add the actual value of the manufacturer code here:

`setHardwareAssistManufacturerCodes(...)`

To find the code, you will need to see the raw bytes of your advertisement.  You can turn on debug logging with `beaconManager.setDebug(true)` and it will log it when detected in the foreground.  The two byte code comes right after the FF byte in the advertisement.

fims, 2019-08-12:
Thanks for the fast reply. I've now tried to add the ManufacturerCode, but haven't been able to figure it out. Here is the log entry:

`Processing pdu type 16: 0201060f166afe0206011464f866746570343810093934383532353030393937334434350000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 18`

As I understand it, the manufacturer is `6efe` and the beacon type is `02`. What I find strange is the layout for the parser has `fe6e` as manufacturer: 
`s:0-1=fe6a,m:2-2=02,p:7-7,i:8-15v,d:4-4,d:5-5,d:6-6`

I tried different combinations but the beacon gets still lost. Here is how I do it:
```
        BeaconParser beaconParser = new BeaconParser().setBeaconLayout("s:0-1=fe6a,m:2-2=02,p:7-7,i:8-15v,d:4-4,d:5-5,d:6-6");
        beaconParser.setHardwareAssistManufacturerCodes(new int[]{0x02});

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(beaconParser);
```

davidgyoung, 2019-09-04:
@fims, sorry for the tardy response.  Your comment shows your beacon is based based on a GATT Service advertisement, so there is not manufacturer code.  There must be a different issue.

Can you please look for a log line like this in the background:  `making scan filter for service:`.  The details of what it says at the end may tell me what is wrong with the scan filter it creates, and why it does not match the beacon packet you show.

fims, 2019-09-05:
Absolutly no prolem. Thank you for taking your time to help me. 

Here is the line you requested: 2019-09-05 15:14:06.674 31043-31043/org.altbeacon.beaconreference D/ScanFilterUtils: making scan filter for service: 0000FE6A-0000-1000-8000-00805f9b34fb 0000fe6a-0000-1000-8000-00805f9b34fb

davidgyoung, 2019-09-05:
Thanks @fims.  Sorry, I should have been more clear.  Can you also show the next lines that are output in logcat after this one (for the next one second)?  I also need to see operating system logs (not filtered by your app) because they show the details of the scan filter that actually gets created.

fims, 2019-09-05:
Please find attached the logcat output:
[](url)
[log.txt](https://github.com/AltBeacon/android-beacon-library/files/3580232/log.txt)

davidgyoung, 2019-09-05:
Got it, thank you.  Here is the line I needed to see:

```
2019-09-05 18:12:08.514 6449-6449/org.altbeacon.beaconreference
D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter
[mDeviceName=null, mDeviceAddress=null,
mUuid=0000fe6a-0000-1000-8000-00805f9b34fb,
mUuidMask=ffffffff-ffff-ffff-ffff-ffffffffffff, mServiceDataUuid=null,
mServiceData=null, mServiceDataMask=null, mManufacturerId=ffffffff,
mManufacturerData=null, mManufacturerDataMask=null]
```

That line is returned by the operating system and it indicates that a
bluetooth scan filter has been set up to match on an advertisement with a
16-bit UUID value of fe6a.  It looks correct.

So if you have your beacon layout set up as:  `
s:0-1=fe6a,m:2-2=02,p:7-7,i:8-15v,d:4-4,d:5-5,d:6-6*`
And you are able to get detections in the foreground successfully with
packets like this that show the 16 bit uuid as 6afe (little endian byte
order, so it looks reversed)

```

Processing pdu type 16:
0201060f166afe0206011464f866746570343810093934383532353030393937334434350000000000000000000000000000000000000000000000000000
with startIndex: 5, endIndex: 18

```

Then I think the scan filter above **should** match this packet.  Since
scan filters must always be used on Samsung when the screen is off, I would
expect that this would match and you would get detections in the
background.  Yet you say you don't.

Is the above a correct summary?  Can you confirm that you are still getting
detections in the foreground?  If so, I think this eliminates the scan
filter as being the problem and the issue must be something else.

fims, 2019-09-05:
Yes thats a correct summary. I get detections in the foreground but they stop as soon as the screen turns black. If the Screen gets turned on (even without unlocking) the beacons are detected again. 

Does the manufacturer have to be registered or can I use a custom manufacturer? There is a list of company identifiers https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/ which doesn't list the manufacturer code fe6a.

fims, 2019-11-08:
I think this Issue can be closed since it's not a bug of the library. It's caused by the Manufacturer (Samsung) and can't be handled by the library.

closed 2019-11-08 by fims