AltBeacon/android-beacon-library #918: Is there some way of accessing timestamps?

thezealousfool, 2019-09-03:
### Expected behavior

Android's onScanResult and onBatchScanResults callbacks pass in the ScanResult object(s). I can get timestamp using this ScanResult (getTimestampNanos). I could not figure out how I could access the timestamp from the Beacon object.

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-09-04:
When using the library's ranging APIs, the beacons detected during the previous scan cycle (1.1 seconds by default, but configurable) are delivered at the end of the scan cycle.  

There is no timestamp to indicate exactly when an individual beacon packet actually was received by the Android device.  The closes you can do is  record a timestamp of when the didRangeBeaconsInRegion callback was made, this gives you a worst case loss of precision of 1.1 seconds using default settings.

It would certainly be possible to add a more specific timestamp as a new feature, but because multiple beacon packets can be received between each callback, there may need to be multiple timestamp fields (e.g. timeFirstDetectedInCycle, timeLastDetectedInCycle or similar).  If you would like to work on such a feature and prepare a pull request I'd be happy to give you pointers to get started.

thezealousfool, 2019-09-05:
Thanks for the clarification. I am new to these stuff but I am happy to give it a shot.

rlatapy-luna, 2020-03-23:
Hi @davidgyoung, I also need to get the timestamp of the beacon detection. My issue is a bit different: it seems that after some times in sleep (~30min and avoiding deep doze), the range notifier is only called when the device wakes up so we cannot know the real detection time. Does it make sense?
I made the necessary to get the timestamp from the `scanResult` into the Beacon object here: https://github.com/rlatapy-luna/android-beacon-library/tree/feature/getDetectionTime
Your opinion would be welcome 😉

davidgyoung, 2020-03-25:
Hi, @rlatapy-luna. Thank you for this suggestion.  I certainly see where this could be helpful.  A couple of thoughts:

1. There is already similar detection "metadata" in the Beacon class like the mPacketCount field.
2. I believe the mTimestampMs that you added is effectively the time the last packet was detected in the scan cycle.  (If two packets were detected, the timestamp will be for the second one, yes?) I think if this is added to the library, it probably makes sense to have two timestamp fields:  mFirstDetectionTimestamp, mLastDetectionTimestamp.
2. It looks like you have covered most of the places this needs to be added -- I believe the new fields needs to be copied though in the following methods on Beacon.java:

    public void writeToParcel(Parcel out, int flags)
    protected Beacon(Parcel in) 
    protected Beacon(Beacon otherBeacon)

If you are interested in merging this into the library, I would welcome a pull request with the above items added.

rlatapy-luna, 2020-03-25:
Thank you for your advices, I opened https://github.com/AltBeacon/android-beacon-library/pull/956