AltBeacon/android-beacon-library #919: [NoClassDefFoundError] Crash on Start Ranging

JonathanImperato, 2019-09-18:
### Expected behavior
Library starts ranging.

### Actual behavior

>     java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/localbroadcastmanager/content/LocalBroadcastManager;
        at org.altbeacon.beacon.BeaconLocalBroadcastProcessor.unregister(BeaconLocalBroadcastProcessor.java:82)
        at org.altbeacon.beacon.BeaconLocalBroadcastProcessor.register(BeaconLocalBroadcastProcessor.java:74)
        at org.altbeacon.beacon.service.ScanJobScheduler.ensureNotificationProcessorSetup(ScanJobScheduler.java:70)
        at org.altbeacon.beacon.service.ScanJobScheduler.schedule(ScanJobScheduler.java:130)
        at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:85)
        at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:92)
        at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:1008)
        at org.altbeacon.beacon.BeaconManager.startRangingBeaconsInRegion(BeaconManager.java:853)

 
### Steps to reproduce this behavior
Add to the gradle this library 
    implementation 'com.google.android.material:material:1.1.0-alpha9' , version alpha9 and above.
### Mobile device model and OS version
OnePlus 3t Android 9, OnePlus 7 Pro Android 9
### Android Beacon Library version
Tested on latest release (2.16.2) and latest pre-release (2.16.3-beta4@aar)

aigori, 2019-09-18:
`androidx.localbroadcastmanager.content.LocalBroadcastManager` has been deprecated, and is removed from Android 10 (API 29).
You should downgrade`targetSdkVersion` to 28 for workaround, and also add below to your app level `gradle.build`

`implementation 'com.android.support:localbroadcastmanager:28.0.0'`

davidgyoung, 2019-09-18:
I don't think this problem originates from this library.   In library version 2.16.2, the class mentioned in the stack trace does not use the deprecated `androidx.localbroadcastmanager.content.LocalBroadcastManager`  it uses `android.support.v4.content.LocalBroadcastManager`.    You can clearly see that in the souce code here: https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconLocalBroadcastProcessor.java#L31

Are you perhaps using code that dynamically changes the classes used by the library to use androidx variants?  If something else is altering the library's classes, then this project can't do anything about it.

JonathanImperato, 2019-09-18:
> I don't think this problem originates from this library. In library version 2.16.2, the class mentioned in the stack trace does not use the deprecated [...]

Well, I did not do anything that was not expected. The suggested solution

> `androidx.localbroadcastmanager.content.LocalBroadcastManager` has been deprecated, and is removed from Android 10 (API 29).
> You should downgrade`targetSdkVersion` to 28 for workaround, and also add [...]

Seems to work well for me.

davidgyoung, 2019-09-18:
Glad to hear you found a solution, @JonathanImperato

aigori, 2019-09-19:
I found details.
The reason of appearing `androidx` package is just Jetifier problem. It convert `support library` into `androidx` automatically. You can find `android.enableJetifier=true` in your `gradle.properties`.
However, although you set `android.enableJetifier` to `false`, the problem will still occur with `android.support.v4` package.
The real problem is `support library` does not support API 29 or above anymore. This project should migrate to `androidx` in the end.
[Maven Repository](https://mvnrepository.com/artifact/com.android.support/support-v4)
[Error message when you use support library with targetSdkVersion 29](https://stackoverflow.com/q/57334494/1010443)

davidgyoung, 2019-09-19:
@khcpietro , can you please clarify this: 

>> You can find android.enableJetifier=true in your gradle.properties.
However, although you set android.enableJetifier to false, the problem will still occur with android.support.v4

I do not understand this part.

khcpietro, 2019-09-19:
My mistake. I mean **You** as @JonathanImperato.
It just explains why `androidx` package appear.

> I don't think this problem originates from this library. In library version 2.16.2, the class mentioned in the stack trace does not use the deprecated [...]

@JonathanImperato may enabled Jetifier, so `android.support` library changed into `androidx`.

In anyway, both `android.support.v4.content.LocalBroadcastManager` and `androidx.localbroadcastmanager.content.LocalBroadcastManager` are no longer available for API 29+.

davidgyoung, 2019-09-19:
I see.  Well, somehow the [reference app](https://github.com/AltBeacon/android-beacon-library-reference) builds fine on SDK 29 and runs on Android 10 without crashing.  And it has `android.enableJetifier=true`

Why does it not have the same problem?

khcpietro, 2019-09-19:
I admit `androidx.localbroadcastmanager.content.LocalBroadcastManager` is not deprecated in Android SDK 29. My apology.

It's problem of `androidx`. In fact, `androidx 1.0.0` == `android.support 28` [link](https://developer.android.com/jetpack/androidx/migrate#prerequisites)
Specifically, `androidx.localbroadcastmanager.content.LocalBroadcastManager` is deprecated from `androidx 1.1.0-alpha01` [link](https://developer.android.com/jetpack/androidx/releases/localbroadcastmanager#1.1.0-alpha01).
So, you can see crush on your reference app when you upgrade `androidx.appcompat:appcompat:1.0.0` into `androidx.appcompat:appcompat:1.1.0`

In summary,
1) Support Library cannot be used on Android SDK 29+ (you can find another crush if you disable Jetifier although you use `appcompat 1.0.0` on your reference app)
2) `androidx.localbroadcastmanager.content.LocalBroadcastManager` can be used on Android SDK 29+ with only `appcompat 1.0.0`

ofir-bar5, 2019-09-19:
Worked for me @khcpietro 
The issue happened when upgrading to androidx.appcompat:appcompat to **1.1.0**.

I downgraded to androidx.appcompat:appcompat to 1.0.2 and everything works fine now.

Targeting the SDK version (either using API 28 or 29) didn't matter for me.

I tested both on the library version 2.16.2 and 2.16.3

matdev, 2020-02-05:
I also have this crash with the new version of the lib 2.16.4
Does it mean it's not possible to target Android SDK 29 with android-beacon-library ?

jared-hexagon, 2020-03-03:
Any update on this? Suddenly getting crashes with version 2.16.4, targeting Android SDK 28 and tested on Android 7 and 8.

matdev, 2020-03-03:
Yes, check this post: https://github.com/AltBeacon/android-beacon-library/issues/948
The solution is here:
https://stackoverflow.com/a/60141077

Biangom, 2020-04-01:
In build.gradle, You should modify "implementation 'androidx.appcompat:appcompat:1.0.2' "

davidgyoung, 2020-04-01:
I am locking this issue as it has become a grab-bag of comments and requests for assistance.  To be clear, this issue is marked as resolved, because there are a number of known solutions.  The specific solution for you may be different depending on your build setup.  The root issue is that the androidx.appcompat:appcompat are in flux ad new versions come out, and using different versions in combination with other libraries can cause the symptoms described above.

To summarize, please try the following:

1. In build.gradle, if you have a line like this, make sure you use the version shown here or newer: `implementation 'androidx.appcompat:appcompat:1.0.2' `
2. If the above does not help, add the following to your dependencies in build.gradle: `implementation 'androidx.localbroadcastmanager:localbroadcastmanager:1.0.0'`

If you try the above and it doesn't work, please open a question on StackOverflow.com.  If we find we need to update these instructions I will.

closed 2020-04-01 by davidgyoung