AltBeacon/android-beacon-library #924: Beacon monitoring intermittently stops in the background.

ravsingh-inzura, 2019-10-12:
### Expected behavior
Beacon library setup to monitor and range whilst app is in background; 
foregroundServiceScanning enabled.
User may not foreground app potentially for days, monitoring and then ranging should continue to function.

### Actual behavior
Intermittently monitoring will stop. At this point foregrounding the app will not resume monitoring. Beacon detection appears to die and App must be force restarted to start detection.

### Steps to reproduce this behaviour
Difficult to pin down but if you run for extended periods it's guaranteed to stop monitoring.

So a question would be, how onForegrounding the App for example, can you detect that the BeaconManager is "healthy" and active for monitoring and ranging?

Is there a way of probing/pinging BeaconManager to determine whether it's functioning?
If the "ping" returns false is there a recommended way of tearing down BeaconManager and restarting clean?   

### Mobile device model and OS version
Samsung S7 Android 8.0, Samsung S8 Android 9.0, Pixel 3 XL Android 10.0.

### Android Beacon Library version
2.16.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-10-14:
Te short answer is, no, it is not possible to query the library to know if it is functioning to detect bluetooth beacons.  

Why?  The library *always* wants to be monitoring if told to do so.  If it is not, then something has killed the app or blocked bluetooth detections.  The library really can't tell if this is the case -- if it could, it would certainly let you know.

I'd be glad to help find out why specific cases are not working after some time.  To do that, I'd need more specific details of how to reproduce with the reference application.  If you can give me that info, please open a new issue.

closed 2019-10-14 by davidgyoung

ravsingh-inzura, 2019-10-15:
Totally understand that, thanks. And when it's backgrounded that makes sense.

The issue here, and hence why the question was raised. If the App is foregrounded, hence very much alive, and we still get no Beacon detection. Is there a way of probing the Beacon to say are you still able to scan, if not then "self heal by rebooting yourself"

The situation we see now is that Beacon detection stops and there is no expectation of it starting again, until the user force restarts the app. 

Even if we had to say for example, we haven't seen a Beacon for 24 hours then 'reboot BeaconManager". Is there a recommended way to tear down and restart BeaconManager clean?

davidgyoung, 2019-10-15:
I have not seen a case like you describe where the library gets into a bad state and stopping and restarting it helps.  (That doesn't mean there is no such case, but I doubt one exists.).  You can certainly unbind the BeaconManager and then later rebind, but I highly doubt this will help.

The most likely caused of what you are seeing are:. (1) The OS has blocked BLE detections by policy.  (2) The Bluetooth stack had gotten into a bad state and must be restarted. (3)
The app itself has gotten itself into a state where it cannot process detections from the library.  There are lots of specific workarounds for (1), and there is a BluetoothMedic class for (2).

ravsingh-inzura, 2019-10-15:
Thanks for the quick response.
We will look into BluetoothMedic, as the BT stack getting into a bad state looks like a viable culprit.