AltBeacon/android-beacon-library #925: Beacon monitoring intermittently stops in the background.

ravsingh-inzura, 2019-10-15:
Expected behavior
Beacon library setup to monitor and range whilst app is in background;
foregroundServiceScanning enabled.
User may not foreground app potentially for days, monitoring and then ranging should continue to function.

Actual behavior
Intermittently monitoring will stop. At this point foregrounding the app will not resume monitoring. Beacon detection appears to die and App must be force restarted to start detection.

#924 

Steps to reproduce this behaviour
Difficult to pin down but if you run for extended periods it's guaranteed to stop monitoring.

Thanks for your feedback in issue #924.

Totally understand that, thanks. And when it's backgrounded that makes sense.

The issue here, and hence why the question was raised. If the App is foregrounded, hence very much alive, and we still get no Beacon detection. Is there a way of probing BeaconManager to say are you still able to scan, if not then "self heal by rebooting yourself"

The situation we see now is that Beacon detection stops and there is no expectation of it starting again, until the user force restarts the app.

Even if we had to say for example, we haven't seen a Beacon for 24 hours then 'reboot BeaconManager". 

**Sorry if this question should be submitted on StackOverflow, but 
is there a recommended way to tear down and restart BeaconManager clean?**

Mobile device model and OS version
Samsung S7 Android 8.0, Samsung S8 Android 9.0, Pixel 3 XL Android 10.0.

Android Beacon Library version
2.16.3

IMPORTANT: This forum is reserved for feature requests or reproducible bugs with the library itself. If you need help with using the library with your project, please open a new question on StackOverflow.com.

androidfu, 2019-10-15:
I too have experienced this behavior.  The logs look normal as everything appears to continue periodic scanning but no beacons are detected until you force close and restart the app.

davidgyoung, 2019-10-15:
@ravsingh-inzura, this issue is just too broad to be actionable.  I have not seen this same behavior on the devices you mention on the reference app.  Have you?

What I need are much more detailed specifics about how to reproduce on a single device model with the reference application -- not a custom app.  Understand that we cannot use this forum to debug custom apps as the number of unknown variables are infinite.

Please try and gather more info starting with the reference app on a single device model.  If you can reproduce the problem, try to reproduce more than once and document the steps to do so.  Also, please look at the logs output by the library in LogCat (possibly setting beaconManager.setDebug(true) for more detail).  If you can find a reproducible test case and attach logs of when detections stop,  this may be actionable.

paulpv, 2020-01-27:
Close as [possibly] fixed by #941 ?