AltBeacon/android-beacon-library #928: didEnterRegion and didExitRegion not getting called when application was killed

Whitelotuscorp, 2019-10-18:
### Expected behavior

didEnterRegion and didExitRegion need to call when application is killed

### Actual behavior

didEnterRegion and didExitRegion getting called when application is killed only in Samsung and Motorola devices with OS version 9 (Pie)
 
### Steps to reproduce this behavior
Install the application in any redmi Xiaomi device after run kill the application entry exit will not be called
### Mobile device model and OS version
Redmi Y3 , OS Version 9 PKQ1.181021.001

### Android Beacon Library version
v 2.16.1 

When am running this library with use of BeaconConsumer
App is running as expected only on samsung devices with any os vesrion

But when m trying to run application in any Xiomi devices with os version 9 or any
It is working as expected when the application is in foreground or in background.

But when m kill the application it will stop working no  didEnterRegion and didExitRegion will be called
 
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2019-10-18:
@Whitelotuscorp, 

There are real challenges getting background detection to work on Xiaomi/Redmi devices due to the manufacturer's undocumented non-standard Android modifications to background task killing.  Read more here:

http://www.davidgyoungtech.com/2019/04/30/the-rise-of-the-nasty-forks
https://dontkillmyapp.com/xiaomi

That said, I would appreciate any help in documenting the specific issues experienced with Xiaomi/Redmi devices as I do not have access to any myself.  Depending on the specific issues, workarounds may be possible.

To address the specifics of this request, I'd appreciate it if you could collect some logs.

1. Verify your app does not use a foreground service or disable scheduled scan jobs by the library.
2. Set up debug level logging in your app by setting beaconManager.setDebug(true);
3. Set up a LogCat capture to filter for your application.
4. Start up your app and turn on a beacon.  Verify logs say it is detected.
5. Turn off your beacon.  Wait 60 seconds.
6. Kill the app
7. Turn on the beacon
8. Continue capturing logs for 15 minutes.  Look for any log messages coming from the library.

If you do not see any messages in step (8), then there is one other test you can run:

1. Install my stopped app detector app here: https://github.com/davidgyoung/stopped-app-detector
2. After killing your app in the task switcher, see if you app gets added to the list of stopped apps. 

See a related reports about Xiaomi/Redmi detection issues here:

https://github.com/AltBeacon/android-beacon-library/issues/865
https://github.com/AltBeacon/android-beacon-library/issues/909

Do you see detections in the background if the app is not killed?  The report in #909 says this does not work either, although that was reported with a Redmi OS version 7.x device.

Whitelotuscorp, 2019-10-18:
@davidgyoung 
Yes I can see detections in the background if the app is not killed

It is not working when app getting killed

And from 8 steps when m kill the application logs are stop getting print

davidgyoung, 2019-10-18:
@Whitelotuscorp, please be sure that you have done tests exactly as I mention.

Using default settings, It might take ~15 minutes (actually 25 minutes worst-case scenario) for the library to restart a bluetooth scan.  If you have not waited 25  minutes for log lines to show up, then it may just be that not enough time has elapsed.  If that really is the case, please use the Stopped App Detector to see if Xiaomi blacklists apps from running after killing with the task switcher.

Whitelotuscorp, 2019-10-18:
@davidgyoung 
I m Using BeaconConsumer
And it is very fast does not wait for 15 minutes

davidgyoung, 2019-10-18:
In the foreground, detections with default settings will be very fast, much less than 15 minutes.  In the background, however, the library with default settings relies on two techniques with Android 8+:

1. A hardware scan filter delivered by intent.  This will also give detections very fast.
2. A scan scheduled by the JobScheduler which runs every ~15 minutes (guaranteed by Android to be 10-25 minutes)

In the specific case you are testing where the app is killed it is technique (2) that will be active.  This is why you must wait up to 25 minutes.

Whitelotuscorp, 2019-10-21:
@davidgyoung 
I waited for 25 minutes, 30 Minutes
But no notification

davidgyoung, 2019-10-21:
@Whitelotuscorp , can you please try installing the [stopped app detector](https://github.com/davidgyoung/stopped-app-detector) and see if your app is shown in the list of stopped apps after being killed from the task switcher?

Whitelotuscorp, 2019-11-14:
@davidgyoung 

I found another issue in this library
When internet is turned off and app was killed then DidEnter and DidExit methods does not get called