AltBeacon/android-beacon-library #929: ConcurrentModificationException in ScanJob line 81

matdev, 2019-10-24:
I posted this issue initially on StackOverFlow:
https://stackoverflow.com/questions/58536934/concurrentmodificationexception-in-scanjob-from-android-beacon-library/58541877

### Expected behavior
No crash

### Actual behavior
I've an app scanning for BLE devices. In Crashlytics, I noticed the following crash:

```
Fatal Exception: java.util.ConcurrentModificationException
       at java.util.ArrayList$Itr.next + 860(ArrayList.java:860)
       at org.altbeacon.beacon.service.ScanJob$1.run + 81(ScanJob.java:81)
       at java.lang.Thread.run + 764(Thread.java:764)

pool-27-thread-2
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$ + 2137(Thread.java:2137)
       at sun.misc.Unsafe.park + 358(Unsafe.java:358)
       at java.util.concurrent.locks.LockSupport.park + 190(LockSupport.java:190)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await + 2059(AbstractQueuedSynchronizer.java:2059)
       at java.util.concurrent.LinkedBlockingQueue.take + 442(LinkedBlockingQueue.java:442)
       at java.util.concurrent.ThreadPoolExecutor.getTask + 1087(ThreadPoolExecutor.java:1087)
       at java.util.concurrent.ThreadPoolExecutor.runWorker + 1147(ThreadPoolExecutor.java:1147)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run + 636(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run + 764(Thread.java:764)

ParseCommandCache.runLoop()
       at java.lang.Object.wait(Object.java)
       at com.parse.ParseCommandCache.runLoop + 664(ParseCommandCache.java:664)
       at com.parse.ParseCommandCache.access$000 + 40(ParseCommandCache.java:40)
       at com.parse.ParseCommandCache$2.run + 188(ParseCommandCache.java:188)

```
### Steps to reproduce this behavior

### Mobile device model and OS version
The crash seems to only occurs on device running Android 8+

### Android Beacon Library version
Using version 2.16.3

davidgyoung, 2019-10-24:
Thanks for the report.  I will work on the fix.

matdev, 2019-12-04:
> Thanks for the report. I will work on the fix.

Hello, any idea when will the fix be released please ?

matdev, 2020-01-21:
Testing the beta version now !

davidgyoung, 2020-01-26:
This is now in a 2.16.4 release

closed 2020-01-26 by davidgyoung