AltBeacon/android-beacon-library #932: Crash when in background

Nicolo-sudo, 2019-11-19:
Hi there!

I'm developing an app with the AltBeacon Library but unfortunately I'm having a big problem.
My app now is fully working only in foreground. The code wich I'm using is the same of the reference application but with some modifications:

> beaconManager.setBackgroundScanPeriod(1100l);
> beaconManager.setBackgroundBetweenScanPeriod(30000l);

My problem is that when the app enters in background and the library found a beacon in a region, the app crashes. In particular, the os shows "The app is not responding" popup.
I suppose the problem is during the "didEnteredInRegion" and "didExitRegion" while the app is in background mode but I'm not sure.

Else if I open the app the "didEnteredInRegion" and "didExitRegion" are working properly without any problems.

Can someone help me to resolve this issue? Thanks in advance!

davidgyoung, 2019-11-19:
I suspect this is an issue with the specific code inside the didEnterRegion/didExitRegion callbacks.   I'd suggest you post that code on StackOverflow.com where I would be glad to help you troubleshoot

closed 2019-11-19 by davidgyoung

Nicolo-sudo, 2019-11-20:
Hi, I have posted a question here:

https://stackoverflow.com/questions/58934627/altbeacon-application-crashes-while-in-background

If you have time, please check it out

Thanks.