AltBeacon/android-beacon-library #933: Background Scans on Android9 Samsung Devices

camaira, 2019-12-03:
Hello and first of all, thank you all very much for this library.

### Expected behavior
I use a foreground service for scanning for beacons, with a notification. I am using BLE Beacons as a Geofence. Fortunately I have many devices to test. The application works perfectly fine with Pixel3 (Android Q), Xiaomi Mi9SE (Android P), Oneplus 7 (Android P), Xiaomi Pocophone F1 (Android P), Motorola G6 (Android O). 

### Actual behavior
The beacon scan on Samsung Galaxy S10 and Galaxy Note 10 works fine if the screen is on. However the scan stops immediately if I turn the screen off, only to resume if it turns on again, even if the device is locked. Those two are the only phones where I see this behaviour.

### Steps to reproduce this behavior
The same problem occurs with the beacon reference application. So by uncommenting the code for the foregroundservice, you can reproduce this issue using a Galaxy S10 or Note 10 and trying to turn the screen off - the phone won't receive beacons any more.

setDebug(true) shows the following when in background:

`D/IntentHandler: got ranging data
D/CycledLeScanner: Waiting to stop scan cycle for another 97 milliseconds
D/CycledLeScanner: Done with scan cycle
D/ScanHelper: Beacon simulator not enabled
D/ScanHelper: Calling ranging callback
D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
    starting a new scan cycle
    We are already scanning and have been for 54231 millis
D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
    Scan started`

### Mobile device model and OS version
Test device is a Samsung Galaxy Note 10, Android 9.

### Android Beacon Library version
Latest, 2.16.3

I highly appreciate any help. Thank you in advance!

davidgyoung, 2019-12-03:
@camaira, Samsung made a custom change requiring that any beacon scan with the screen off have a non-empty scan filter.  The library was modified to accommodate this change, but for it to work the beacon you are using must match the scan filter the library generates.  You can read more about this here:

https://github.com/AltBeacon/android-beacon-library/issues/769

I suspect the beacon you are using may not be matching the Samsung-only scan filter used in the background.  This might happen if the beacon is uses a manufacturer advertisement (iBeacon, AltBeacon) but has a non-standard manufacturer identifier.

With debug on, can you please post the full bytes of the advertisement when the beacon is actually detected?

camaira, 2019-12-05:
Thank you very much for your answer! Indeed, I use a manufacturer specific beacon type based on Altbeacon, but with a slightly altered beacon layout; the manufacturer ID is the one of my company (it is an official Bluetooth SIG manufacturer ID indeed - 0x0485), and it is complete custom hardware sending the beacon.
I am afraid I cannot post advertisement data here due to a NDA with my company... But I'll try a different manufacturer ID and let you know if there is a change! 

Thank you again!

davidgyoung, 2019-12-05:
Ok good.  You can easily fix the problem by setting your manufacturer code with this method:

See this method on BeaconParser here:

https://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconParser.html#setHardwareAssistManufacturerCodes-int:A-

The default list of manufacturer codes is set to 0x004c (apple) for all beacon formats except AltBeacon. This is generally what you want, unless you find that your particular iBeacon is using a non-standard manufacturer code.

closed 2019-12-05 by davidgyoung

paulpv, 2019-12-19:
FYI, I suspect this is not actually fixed:
https://stackoverflow.com/a/57654627/252308

> I faced the same issue. I had Scan filters in order to scan BLE devices even if the screen were locked. But on Samsung devices it didn't work, so I search on Samsung forum and I discovered Knox SDK (https://seap.samsung.com/sdk/knox-android).
> 
> And it was the solution of my problem. All you have to do is add it to your app, create a license and activate it and finally use this method addPackageToBatteryOptimizationWhiteList to unlock the scan when the Samsung device screen is lock.

If you want to scan BLE on a Samsung device when the screen is locked then you need to apply for and install and initialize/whitelist your app in their Knox SDK.

sba923, 2020-06-02:
> If you want to scan BLE on a Samsung device when the screen is locked then you need to apply for and install and initialize/whitelist your app in their Knox SDK.

Can you please explain what you mean by that? Does one need to contact Samsung for that? Is there any documentation you could point us to?

paulpv, 2020-06-02:
https://docs.samsungknox.com/devref/knox-sdk/reference/com/samsung/android/knox/application/ApplicationPolicy.html#addPackageToBatteryOptimizationWhiteList(com.samsung.android.knox.AppIdentity)

I found the Samsung KNOX SDK documentation to be extremely underwhelming.
It was just a theory that I had based on someone else's experience.
I tried it myself and could not tell if my code had any affect or resulted in any changes...I still could not scan with the screen off without a scan filter.

As @davidgyoung has implied in related issues, Knox SDK may be a red-herring; all you really need is a valid non-empty scan filter set.

Whathecode, 2021-07-09:
> I suspect the beacon you are using may not be matching the Samsung-only scan filter used in the background. This might happen if the beacon is uses a manufacturer advertisement (iBeacon, AltBeacon) but has a non-standard manufacturer identifier.

Excuse my ignorance, but what does this mean exactly? We are using [Feasycom beacons](https://www.feasycom.com/) which allow configuring the manufacturer ID. We were unaware of the impact this could have on beacon detection; in fact, most apps detect the beacon just fine.

In [an older post I read](https://github.com/AltBeacon/android-beacon-library/issues/769#issuecomment-445494234):

> Beacons emitting an AltBeacon advertisement with a manufacturer code other than that of Radius Networks (the default) will not be matched with this scan filter unless the app developer explicitly registers another manufacturer code with the library. That won't affect many apps, but I don't want to break anything if I don't have to.

Setting the manufacturer ID of the beacon to Radius Networks as hinted at here (0x0118) seems to work, but why exactly is Radius Networks "the default"? Would any of the other "standard" IDs work?

Either way. Glad I found this thread. Thanks for the information!

davidgyoung, 2021-07-09:
Radius Networks is the company that invented Altbeacon, so their Bluetooth SIG assigned company ID of 0x118 is usually what is put in AltBeacon manufacturer advertisement headers.  This library was not originally designed to require or enforce a specific manufacturer Id.  In most cases any manufacturer id is allowed.  

But when BLE hardware scan filters are used (which Samsung started requiring two years after this library’s API was designed) the manufacturer Id must be explicitly specified for a match.  The library defaults it to 0x118 which works the vast majority of the time as most all AltBeacons use 0x118 by historic convention.  But as you have noticed with FeasyBeacon if you change this, it breaks this library’s  detections with scan filters.  

I am sorry for the trouble this gave you.  Any thoughts on how to prevent others from running into this same problem or finding the answer more easily?  The library is designed to make it easy for the 99% case.  Unfortunately you are the other 1% (or less).

I suppose the library could detect the case where a non-standard altbeacon manufacturer code is detected (in the foreground without scan filters at least) and put a warning in LogCat.  But my guess is that you would not have noticed that.

Whathecode, 2021-07-09:
> But my guess is that you would not have noticed that.

Indeed; I would not have. In fact, I prefer runtime exceptions in cases like this because this way problems are caught much earlier. Fail early and fail often. :)

> Any thoughts on how to prevent others from running into this same problem or finding the answer more easily?

Without having an understanding of how the underlying APIs have evolved over the years, and thus risking suggesting a naïve solution:
I think it would make sense to expose `manufacturerId` as part of the `Region` configuration. It is related to [the other suggestion I made in regards to `Region` in the API](https://github.com/AltBeacon/android-beacon-library/issues/1037). The problem was it was hidden to me (as an API user) that a scan would happen for devices with settings which are out of my control. I don't think a static method through `BeaconParser` is a suitable solution. Global state brings all the problems of when should I call this, who else calls this, etc. In my specific case, I wouldn't have run into the issue if I could've just specified it in `Region`.

A default value in the API could then be the `Radius Networks` constant, resulting in more self-documenting code. If I understand correctly, the problem then is how can you scan for _any_ manufacturer ID. If manufacturer ID is optional, implying a default fallback will be used for hacks such as the Samsung background scanning fallback, I'd at the very least document that on this optional parameter (i.e., wildcard might not work for all beacons while scanning when the display is turned off on Samsung devices). But, even better would likely involve a separate API which returns `false` on Samsung devices (e.g., `scanSupportedOnDevice`) which throws an error in logcat if you are trying to scan on an unsupported device; again, this would require incorporating `manufacturer ID` in the scan, which overall seems like a good idea to model this one way or the other.

davidgyoung, 2021-07-09:
Yeah, I originally had the same thought about the runtime exception.  The problem is that an app might come across an beacon with a non-standard manufacturer id in the wild.  This would often cause unexpected crashes for end users  after app release, which would be awful.  This could often happen even if the developer never saw such a crash :(

I think your other suggestions are good.  The trick is balancing any change vs:

1. Keeping the API simple for the 99% use case (iBeacon only always Apple manufacturer id of 0x004c)
2. Backward compatibility of the API to prevent a breaking change

Whathecode, 2021-07-09:
> Backward compatibility of the API to prevent a breaking change

A `RegionBuilder` could maintain backwards compatibility most likely. You can then overload the `scan` calls with a builder vs. explicit `Region` and maintain the same old APIs, maybe apply `@deprecated` to encourage people to start using the builder.

The codebase could probably be cleaned up considerable using polymorphism instead, but it could be a nice transition face as you phase out the old API.