AltBeacon/android-beacon-library #935: App crashes immediately after starting when using bootstrapnotifier on Huawei phones after EMUI update to 9.1

arickatz, 2019-12-11:
### Continue working as before the EMUI update

### crash on start

### use  bootstrapnotifier on Huawei phones after EMUI update to 9.1

### Huawei phones after EMUI update to 9.1

### 2.16.3

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

bloushi, 2019-12-11:
We are facing the same issue of immediate crashes with a Huawei Mate20 X. Here is a stack trace of the issue we are facing:
`12-11 15:18:34.832: E/AndroidRuntime(8250): FATAL EXCEPTION: queued-work-looper
12-11 15:18:34.832: E/AndroidRuntime(8250): Process: com.ethos.indigo.beta, PID: 8250
12-11 15:18:34.832: E/AndroidRuntime(8250): java.lang.UnsatisfiedLinkError: No implementation found for android.content.Context md549f4a3f06cba147d1b884263aa8ffcd6.IndigoAndroidApplication.n_getApplicationContext() (tried Java_md549f4a3f06cba147d1b884263aa8ffcd6_IndigoAndroidApplication_n_1getApplicationContext and Java_md549f4a3f06cba147d1b884263aa8ffcd6_IndigoAndroidApplication_n_1getApplicationContext__)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at md549f4a3f06cba147d1b884263aa8ffcd6.IndigoAndroidApplication.n_getApplicationContext(Native Method)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at md549f4a3f06cba147d1b884263aa8ffcd6.IndigoAndroidApplication.getApplicationContext(IndigoAndroidApplication.java:39)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at android.rms.iaware.AppSceneRecogManager.init(AppSceneRecogManager.java:157)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at huawei.android.app.HwActivityThreadImpl.reportBindApplicationToAware(HwActivityThreadImpl.java:136)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at android.app.ActivityThread$PreloadThreadHandler.handleMessage(ActivityThread.java:1871)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at android.os.Handler.dispatchMessage(Handler.java:112)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at android.os.Looper.loop(Looper.java:216)
12-11 15:18:34.832: E/AndroidRuntime(8250):     at android.os.HandlerThread.run(HandlerThread.java:65)
12-11 15:18:34.836: W/monodroid(8250): Creating public update directory: `/data/user/0/com.ethos.indigo.beta/files/.__override__`
12-11 15:18:34.836: W/monodroid(8250): Using override path: /data/user/0/com.ethos.indigo.beta/files/.__override__
12-11 15:18:34.836: W/monodroid(8250): Using override path: /storage/emulated/0/Android/data/com.ethos.indigo.beta/files/.__override__
12-11 15:18:34.836: W/monodroid(8250): Using runtime path: /data/app/Mono.Android.DebugRuntime-VB-Ewj9MyGBIoCRh783XrA==/lib/arm64
12-11 15:18:34.836: W/monodroid(8250): checking directory: `/data/user/0/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): directory does not exist: `/data/user/0/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): checking directory: `/storage/emulated/0/Android/data/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): directory does not exist: `/storage/emulated/0/Android/data/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): checking directory: `/storage/emulated/0/../legacy/Android/data/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): directory does not exist: `/storage/emulated/0/../legacy/Android/data/com.ethos.indigo.beta/files/.__override__/lib`
12-11 15:18:34.836: W/monodroid(8250): Trying to load sgen from: /data/user/0/com.ethos.indigo.beta/files/.__override__/libmonosgen-2.0.so
12-11 15:18:34.836: W/monodroid(8250): Trying to load sgen from: /storage/emulated/0/Android/data/com.ethos.indigo.beta/files/.__override__/libmonosgen-2.0.so
12-11 15:18:34.836: W/monodroid(8250): Trying to load sgen from: /storage/emulated/0/../legacy/Android/data/com.ethos.indigo.beta/files/.__override__/libmonosgen-2.0.so
12-11 15:18:34.836: W/monodroid(8250): Trying to load sgen from: /data/app/com.ethos.indigo.beta-0pNBtvNcyeFiLZKUjj3F0g==/lib/arm64/libmonosgen-2.0.so
12-11 15:18:34.837: W/monodroid(8250): symlink exists, recreating: /data/user/0/com.ethos.indigo.beta/files/.__override__/links/libmonosgen-2.0.so -> /data/app/Mono.Android.DebugRuntime-VB-Ewj9MyGBIoCRh783XrA==/lib/arm64/libmonosgen-64bit-2.0.so
12-11 15:18:34.837: W/monodroid(8250): Trying to load sgen from: /data/user/0/com.ethos.indigo.beta/files/.__override__/links/libmonosgen-2.0.so
12-11 15:18:34.839: W/monodroid-debug(8250): Trying to initialize the debugger with options: --debugger-agent=transport=dt_socket,loglevel=0,address=127.0.0.1:29338,server=y,embedding=1`

The OS version is 9.1.1 and AltBeacon version we are using is 2.15.0

davidgyoung, 2019-12-11:
I don:t see anything in that stacktrace that comes from this library.  How do you know the crash is related to the library?

atifgulzar1981, 2019-12-12:
I am facing the same problem, when I remove "IBootstrapNotifier" implementation, the app starts working fine. The error is same which @bloushi shared.

bloushi, 2019-12-16:
The only way we found to make this work is if we move the BootstrapNotifier implementation to the mainActivity instead of class that extends android.app.Application.

arickatz, 2019-12-16:
I was able to make it work, if I do a type of late binding to the IBootstrapNotifier from the application class, meaning if create a different class that inherits from the IBootstrapNotifier implement the overrrides
and from the oncreate of the application class call the new class and reference the interface and transfer that interface to the RegionBootstrap object.
This works for all phones, yet on the Huawei and OnePlusX the ibeacon calls do not wake a killed app.
This might be due to the EMUI killing non permitted background schedulers \ timers