AltBeacon/android-beacon-library #937: didRangeBeaconsInRegion called faster than scanning interval

atotalnoob, 2019-12-30:
### Expected behavior
didRangeBeaconsInRegion only is called after the scanning interval elapses. IE with these intervals:

```kotlin
        beaconManager.foregroundScanPeriod = 2111
        beaconManager.foregroundBetweenScanPeriod = 0
        beaconManager.updateScanPeriods()
```
didRangeBeaconsInRegion is called no more than once every `2111ms`

### Actual behavior
didRangeBeaconsInRegion is called more than once per scanning interval. We take one timestamp at the beginning of the callback and then utilize that timestamp for all of our beaconLogs. Once the beaconLogs have all been created, they are sent to our API. 

```kotlin
override fun didRangeBeaconsInRegion(beacons: Collection<Beacon>, region: Region) {
        if (beacons.isEmpty()) {
            return
        }

        val readingTakenAt = Date().time
        val someList = ArrayList<BeaconLog>()
        for (beacon in beacons){
           someList.add(BeaconLog(beacon))
        }

        // business logic and send to api
}
```

Querying our database revealed that, the frequency being divisible by 2000 meant it was more likely to be less than the scanning interval. 

```
broadcast_frequency,minimum (seconds)
250ms,0.254
500ms,0.358
750ms,2.108
1000ms,0.267
1250ms,2.115
1500ms,2.108
1750ms,2.014
2000ms,0.521
```

We changed our interval from `2100` to `2111` (prime number) and repeated the experiment.

```
broadcast_frequency,minimum (seconds)
250ms,1.004
500ms,1.004
750ms,2.108
1000ms,1.004
1250ms,2.115
1500ms,2.108
1750ms,2.014
2000ms,1.437
```

I would share the logs, but this seems to be reproducible across devices, given enough time, and is difficult to gather logs for.

### Steps to reproduce this behavior

Using the sample application, set up foreground service, put beacons at various broadcasting frequencies nearby the device, (<1m) let scan for 24 hours, calculate minimum time between broadcast. 

### Mobile device model and OS version

`Nokia 4.2 TA-1133` - Android 9
`Samsung T-380` - Android 9
`Pixel 3a` - Android 10
`Nexus 9` - Android 5,6,7

### Android Beacon Library version

`implementation 'org.altbeacon:android-beacon-library:2.16.3-beta4@aar'`

atotalnoob, 2020-01-02:
Is this related to #198?

davidgyoung, 2020-01-03:
It's hard to say for sure why this might happen without seeing the raw timestamps of when the scans started and ended.  

Two points that may affect your measurements:

1. The library attempts to normalize the scan intervals so they start on a time period that is evenly divisible by the scan period.  This is based on system time on the Android device.
2. If the device CPU is busy (or even if particular threads are busy) it may cause delays in when certain beacon results are delivered.  If one scan result is delayed and the next one is not, it may cause what you report.