AltBeacon/android-beacon-library #938: Distance "Infinity" meters

codeproof-dev, 2020-01-03:
Hello,

I get the distance "Infinity" meters even though beacon is right next to the device.

Debug logcat:
01-02 16:39:44.309 13973-13973/com.codeproof.device.security D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
01-02 16:39:44.316 13973-14090/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 16:39:44.317 13973-14090/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 0201060aff4c0010054b1cc29063000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 13
    This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201060aff4c0010054b1cc29063000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.318 13973-14090/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
    Processing pdu type FF: 02010613ff4c000c0e0076120b48cc97a8fdc604907c22000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 22
01-02 16:39:44.319 13973-14090/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010613ff4c000c0e0076120b48cc97a8fdc604907c22000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.325 13973-14094/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 16:39:44.325 13973-14094/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 03
01-02 16:39:44.326 13973-14091/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 16:39:44.326 13973-14094/com.codeproof.device.security D/BeaconParser: Processing pdu type 16: 0201060303aafe1116aafe20000c448000000038730000e6390e16f0ff1b020902dc0d30000124640d09434f444550524f4f462d44440000000000000000 with startIndex: 9, endIndex: 24
01-02 16:39:44.326 13973-14091/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 02010613ff4c000c0e081957533fa2cc1c9e3ec24858c1000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 22
01-02 16:39:44.326 13973-14094/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201060303aafe1116aafe20000c448000000038730000e6390e16f0ff1b020902dc0d30000124640d09434f444550524f4f462d44440000000000000000
01-02 16:39:44.327 13973-14091/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010613ff4c000c0e081957533fa2cc1c9e3ec24858c1000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.332 13973-14093/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 16:39:44.334 13973-14093/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 02010613ff4c000c0e081957533fa2cc1c9e3ec24858c1000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 22
01-02 16:39:44.335 13973-14093/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010613ff4c000c0e081957533fa2cc1c9e3ec24858c1000000000000000000000000000000000000000000000000000000000000000000000000000000
    Ignoring pdu type 01
01-02 16:39:44.336 13973-14093/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 0201060aff4c001005411c399e3a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 13
    This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201060aff4c001005411c399e3a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.343 13973-14092/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff060001092002ede5a644e4908f33a462dd96bb29b11df5bb43a344b52000000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 16:39:44.344 13973-14092/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff060001092002ede5a644e4908f33a462dd96bb29b11df5bb43a344b52000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.353 13973-14094/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 16:39:44.354 13973-14094/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 0201060aff4c001005411c0bfe12000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 13
    This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201060aff4c001005411c0bfe12000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.358 13973-14091/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff060001092002866fc38ee7df6716026237822e037f95232102e558f9f800000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 16:39:44.361 13973-14091/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff060001092002866fc38ee7df6716026237822e037f95232102e558f9f800000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.365 13973-14093/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff060001092002f9fb8c6f9111dff611078b7bf16b07da163bfee87d377700000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 16:39:44.368 13973-14093/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff060001092002f9fb8c6f9111dff611078b7bf16b07da163bfee87d377700000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.427 13973-14090/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff060001092000b5b9f2a8de660fdea46af1d69c395f2313702187ea825200000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 16:39:44.428 13973-14090/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff060001092000b5b9f2a8de660fdea46af1d69c395f2313702187ea825200000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.476 13973-14092/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff06000109200238d40ee37e4dfe407f707da03a367586c4b60faf78c68100000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
    This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff06000109200238d40ee37e4dfe407f707da03a367586c4b60faf78c68100000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.477 13973-14092/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff060001092002083aa75f90b82a1340806c7313d4fdbd0130c6446dc11e00000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 16:39:44.478 13973-14092/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff060001092002083aa75f90b82a1340806c7313d4fdbd0130c6446dc11e00000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.479 13973-14092/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff0600010920023b05d480ba03a726314f6d8720e224ad3ff748b4c7c1b700000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
    This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff0600010920023b05d480ba03a726314f6d8720e224ad3ff748b4c7c1b700000000000000000000000000000000000000000000000000000000000000
01-02 16:39:44.480 13973-14092/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
[... cut ...]
01-02 16:39:53.264 13973-13973/com.codeproof.device.security D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
01-02 16:39:54.372 13973-13973/com.codeproof.device.security D/ScanHelper: Beacon simulator not enabled
01-02 16:39:54.384 13973-13973/com.codeproof.device.security D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
01-02 16:39:55.490 13973-13973/com.codeproof.device.security D/ScanHelper: Beacon simulator not enabled

davidgyoung, 2020-01-03:
The problem is that your beacon is transmitting a measured power value of 00, indicating that the RSSI should be an impossibly strong 0dBm at a distance of 1 meter.  No beacon in the world has such a powerful transmitter -- a typical measured power value at 1 meter is -59 dBm.  So since the library measures a much, much weaker signal level, it correctly calculates that the distance must be *extremely* far away.  This incorrect result is caused by that incorrect measured power value.

The solution is to calibrate your beacon and  configure it to transmit a correct measured power value.  Please consult your beacon manufacturer instructions on how to do this, and if you need more help, please open a question on StackOverflow.com.

closed 2020-01-03 by davidgyoung

codeproof-dev, 2020-01-03:
i set the TxPower to 5 dBm and the results are same.  Our Beacon supports -21 dBm to 5 dBm.  The library always returns distance as "infinity" meters. I suspect this is a bug in the distance calculation code. 

Beacon TxPower spec is below:

[image]

davidgyoung, 2020-01-04:
@codeproof-dev, regardless of what that table says, the Tx Power level shown is most certainly not the same thing as the measured power at one meter -- it is probably using a different reference point.  

Again, the correct way to do this is to *calibrate* your beacon using an app like [BeaconScope](https://play.google.com/store/apps/details?id=com.davidgyoungtech.beaconscanner&hl=en_US), which has the feature built-in.  You simply hold the phone one meter away and run the calibration, then use the value you get from the calibration to configure your beacon to transmit that measured power value as part of the advertisement. How to do this for your specific beacon must come from the manufacturer.

Please understand that this forum is reserved for bug reports and feature requests for the library itself.  This is not the proper forum for assistance in configuring a specific manufacturer's beacon for use with the library, which is why I closed this issue.  For that kind of assistance, I am happy to help further if you open a question on StackOverflow.com.