AltBeacon/android-beacon-library #939: This is not a matching Beacon advertisement. (Was expecting be ac.

codeproof-dev, 2020-01-03:
Why i am seeing this?  Do i need to set the setBeaconLayout? 

```
01-02 21:44:54.510 8302-8478/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 01
01-02 21:44:54.513 8302-8478/com.codeproof.device.security D/BeaconParser: Ignoring pdu type 0A
01-02 21:44:54.525 8302-8478/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 02011a020a0c0aff4c001005511cb5a87a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 8, endIndex: 16
01-02 21:44:54.536 8302-8478/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02011a020a0c0aff4c001005511cb5a87a000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
01-02 21:44:54.583 8302-8479/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff0600010920025f0c64e3b65323db96333267b760a2e20d6ef37741833500000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 21:44:54.596 8302-8479/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff0600010920025f0c64e3b65323db96333267b760a2e20d6ef37741833500000000000000000000000000000000000000000000000000000000000000
01-02 21:44:54.687 8302-8480/com.codeproof.device.security D/BeaconParser: Processing pdu type FF: 1eff0600010920025f0c64e3b65323db96333267b760a2e20d6ef37741833500000000000000000000000000000000000000000000000000000000000000 with startIndex: 2, endIndex: 30
01-02 21:44:54.700 8302-8480/com.codeproof.device.security D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 1eff0600010920025f0c64e3b65323db96333267b760a2e20d6ef37741833500000000000000000000000000000000000000000000000000000000000000
```

davidgyoung, 2020-01-03:
The log lines are normal, and indicate that you only have a layout set to detect AltBeacon packets, but that all the packets detected in the log snippet are Apple non-beacon packets.  In other words, no beacon packets are visible when the log was taken.

If you want to detect iBeacon you will need to add a special layout (a Google search will show how to do this) and you need to make sure there is an iBeacon transmitter in the vicinity.

For more help, please open a question on StackOverflow.com.

closed 2020-01-03 by davidgyoung

codeproof-dev, 2020-01-03:
Our also beacon transmits iBeacon and EddyStone. So i need to set the parser for these. Thanks!