AltBeacon/android-beacon-library #945: didRangeBeaconsInRegion method giving empty beacons list even when beacons are present in region

satishkumar-tudip, 2020-01-30:
### Expected behavior
Details:
 - Need to get the Beacons list in the Collection<Beacon> Parameter in didRangeBeaconsInRegion method when the beacons are present within the Region.

### Actual behavior
 Details:
 - I am scanning for particular beacon UUID and my code for scanning the beacon is:
           **Region region = new Region("myRangingUniqueId", Identifier.parse(mBeaconUUID), null, null);
           mBeaconManager.startRangingBeaconsInRegion(region);**** 
- The time that I spent on scanning the beacons is 5 secs. within 5 secs if the beacon is not connected then we need to show the not connected message else we need to connect with the beacon.
- I am receiving onBeaconServiceConnect callback in every 1 second but the beacons list in the callback is 0.
   The code for onBeaconServiceConnect is
        
   **@Override
    public void onBeaconServiceConnect() {
        mBeaconManager.setRangeNotifier(new RangeNotifier() {
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                if (beacons.size() > 0) {
                    
                } 
            }
        });
    }**
   Since the setRangeNotifier callback is deprecated i tried using the addRangeNotifier but the behaviour is same.
Before adding the RangeNotifier I have removed all Notifiers using the removeAllRangeNotifiers but also the behavior is the same.
 
- I have a search around the internet and code is the same. Can you please let me know the issue and why we are not getting the Beacons in the Callback?

### Steps to reproduce this behavior

 - Try to connect to the beacon in 5 secs bypassing the Beacon UUID in the Region.
 - Check if you are getting the list in the onBeaconServiceConnect callback.

### Mobile device model and OS version
 I have tried in Couple of Devices but it is giving the same behavior in all:
  - Samsung SM-A305F (Version 9)
  - Redmi K20 (Version 9)
  - Realme 2 (Version 9)
  - Samsung on5 Pro (Versio 7)

### Android Beacon Library version
  implementation 'org.altbeacon:android-beacon-library:2+'

Here are the attached Logs:
[image]

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-01-30:
This is the most common problem people face when first using the library.  Several things must be in place for detections to work.   Here is an answer I wrote on StackOverflow.  https://stackoverflow.com/a/48583330/1461050  If you try those steps and find it still doesn't work, please open a new question on StackOverflow.com with details on what you tried, and I will help more there.

closed 2020-01-30 by davidgyoung

satishkumar-tudip, 2020-02-03:
Hi David,

I have gone through the link which you have shared and I have verified all the steps which you have mentioned. I have installed the BeaconSocpe app and tried scanning the Beacons at same time in different mobiles in BeaconScope and it is scanning the list in the Redmi K20, Realme 2 Pro and I am unable to get the Beacons list in Samsung SM-A305F. All mobiles are of Android Version 9 and same app is not giving Beacons list in some Mobiles at the same time. 
Whenever I am getting the Beacons list in the BeaconScope I am able to get the Beacons list in my App also.

[BeaconVideo.zip](https://github.com/AltBeacon/android-beacon-library/files/4147338/BeaconVideo.zip)

Can you please verify the Os issue which is occurring in some Mobiles?