AltBeacon/android-beacon-library #946: If you are using BOSE bluetooth headphones, a ranging delay is detected while the music is on.

Bibi-byte, 2020-01-31:
### Expected behavior
While listening to music with BOSE bluetooth headphones, the scanning of the beacons 
keep on.

### Actual behavior
The scan is set every 1100ms, but the scan of the beacons while listening to music has a delay of 20 30 seconds.

### Steps to reproduce this behavior

- The application starts.
- Application in background.
- Connect BOSE Bluetooth headphones.
- Youtube or Spotify starts
- Read the logs
### Mobile device model and OS version
Samsung S9 
### Android Beacon Library version
2.16.1

Hi!
I have done many tests with different devices without problems. But Samsung continues to have delays in the scan.
Thanks!

davidgyoung, 2020-01-31:
I suspect this is not necessarily specific to BOSE bluetooth headphones, but is instead a general problem on the Samsung S9 (and perhaps related devices) whenever any bluetooth headset is connected and active.  If so, this indicates either a hardware issue or an issue with the bluetooth stack on the device.  While it is worthwhile documenting the resulting behavior for the sake of other people who encounter this, I don't think it is anything this library can do to solve these problems.

Bibi-byte, 2020-02-06:
Hi! I have tested others headphones, but delay only happens with BOSE bluetooth headphones more... For others, the delays are 3-4 seconds for BOSE, 20-30 seconds.
Could it be their combination?

davidgyoung, 2020-02-06:
What specific model of headphones is this?  Does it specify it uses a
Bluetooth LE audio profile vs. Classic Bluetooth?

Bibi-byte, 2020-02-06:
BOSE soundSport model 423729, it uses Classic Bluetooth! 

In LogCat I observe that  didDetermineStateForRegion method return OUTSIDE when there is a delay, after 20-30 second return INSIDE and scan begins again.