AltBeacon/android-beacon-library #947: Os issue while scanning the iBeacons using the AltBeacon Library

satishkumar-tudip, 2020-02-04:
Hi David,

I have already raised the ticket 
**https://github.com/AltBeacon/android-beacon-library/issues/945#issuecomment-581341649**. 
I have gone through the link which you have shared and I have verified all the steps which you have mentioned. I have installed the BeaconScope app and tried scanning the Beacons at the same time in different mobiles in BeaconScope and it is scanning the list in the Redmi K20, Realme 2 Pro and I am unable to get the Beacons list in Samsung SM-A305F. All mobiles are of Android Version 9 and the same app is not giving Beacons list in some Mobiles at the same time.
Whenever I am getting the Beacons list in the BeaconScope I am able to get the Beacons list in my App also.

[BeaconVideo.zip](https://github.com/AltBeacon/android-beacon-library/files/4152312/BeaconVideo.zip)

Can you please verify the Os issue which is occurring in some Mobiles?

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

satishkumar-tudip, 2020-02-04:
The main thing is I want to scan for the Beacons only when the app is Foreground. I don't want to Perform Background Scanning. But for foreground scanning also I am not getting the Beacons List every time.

davidgyoung, 2020-02-04:
The inability to detect at all on the Samsung A30 is concerning.  A few possibilities:

* The phone model does not support BLE scanning
* The phone is in a state where such scanning is disabled
* The phone you are testing has a hardware flaw making it unable to scan.

I do not have an A30 to test with.  This is an all volunteer effort so any help investigating the above possibilities would be appreciated.

Anybody else see beacon detections working on an A30?

davidgyoung, 2020-02-04:
It looks like the A30 was at least certified by Bluetooth SIG

https://www.droidshout.com/samsung-galaxy-a30/

Youight try turning off WiFi during Bluetooth scanning to see if that makes it work.  Earlier Android phones like the Nexus 4 and original Moto G could not get Bluetooth scan results with WiFi on.

satishkumar-tudip, 2020-02-05:
Hi David,

I have a doubt regarding the scanning of the Beacons. I have set the Monitoring and ranging at the same time using the below code:

mBeaconRegion = new Region("myRangingUniqueId",
                        Identifier.parse(UUID), null, null);
                mBeaconManager.startMonitoringBeaconsInRegion(mBeaconRegion);
                mBeaconManager.startRangingBeaconsInRegion(mBeaconRegion);

In onBeaconServiceConnect method

mBeaconManager.setMonitorNotifier(new MonitorNotifier() {
            @Override
            public void didEnterRegion(Region region) {
                Log.d("*Beacon", "on Enter");
            }

            @Override
            public void didExitRegion(Region region) {
                Log.d("*Beacon", "on Exit");
            }

            @Override
            public void didDetermineStateForRegion(int i, Region region) {
                Log.d("*Beacon", "on determinate state");
            }
        });

        mBeaconManager.setRangeNotifier(new RangeNotifier() {
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                if (beacons.size() > 0) {
                    Log.d("*Beacon", "size > 0");
                } else {
                    Log.d("*Beacon", "size < 0");
                }
            }
        });

When I have bound the beacon service onBeaconServiceConnect method is called and monitoring and ranging are started.
In Monitoring the didDetermineStateForRegion method is called and the didRangeBeaconsInRegion gets executed every one second. 

I came to know one thing that whenever didEnterRegion gets called after the didDetermineStateForRegion then I am getting the Beacons list otherwise I am getting the Beacon size as zero.

can you please explain why the didEnterRegion is not getting called everytime after the didDetermineStateForRegion? Did I miss something?

satishkumar-tudip, 2020-02-05:
Here are the logs that I captured:

Log 1: (Did not connect to Beacon)
2020-02-05 16:43:47.791 19341-19341/com.app.example D/*Beacon: on Service Connect
2020-02-05 16:43:47.805 19341-19341/com.app.example D/*Beacon: Ranging
2020-02-05 16:43:47.840 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:43:49.104 19341-19341/com.app.example D/*Beacon: size < 0
2020-02-05 16:43:50.208 19341-19341/com.app.example D/*Beacon: size < 0
2020-02-05 16:43:51.311 19341-19341/com.app.example D/*Beacon: size < 0
2020-02-05 16:43:52.413 19341-19341/com.app.example D/*Beacon: size < 0
2020-02-05 16:43:52.964 19341-19341/com.app.example D/*Beacon: unbinding
2020-02-05 16:44:23.067 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:44:23.067 19341-19341/com.app.example D/*Beacon: on Enter

Log2: (Connected)

2020-02-05 16:48:24.014 19341-19341/com.app.example D/*Beacon: on Service Connect
2020-02-05 16:48:24.026 19341-19341/com.app.example D/*Beacon: Ranging
2020-02-05 16:48:24.055 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:48:24.644 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:48:24.644 19341-19341/com.app.example D/*Beacon: on Enter
2020-02-05 16:48:25.309 19341-19341/com.app.example D/*Beacon: size > 0
2020-02-05 16:48:25.313 19341-19341/com.app.example D/*Beacon: unbinding

Log3: (Connected)

2020-02-05 16:48:38.901 19341-19341/com.app.example D/*Beacon: on Service Connect
2020-02-05 16:48:38.913 19341-19341/com.app.example D/*Beacon: Ranging
2020-02-05 16:48:38.939 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:48:39.283 19341-19341/com.app.example D/*Beacon: on determinate state
2020-02-05 16:48:39.283 19341-19341/com.app.example D/*Beacon: on Enter
2020-02-05 16:48:40.177 19341-19341/com.app.example D/*Beacon: size > 0
2020-02-05 16:48:40.182 19341-19341/com.app.example D/*Beacon: unbinding

satishkumar-tudip, 2020-02-06:
Hi 
can I get an update on the issue?

davidgyoung, 2020-02-06:
@satishkumar-tudip, please do not add unrelated  questions on an issue of a different topic.   If you want general help using this library, StackOverflow is the proper forum.   

Please understand that both this project and StackOverflow are free volunteer services so you will have to be patient in getting responses.  If you need a higher level of support, you can consider hiring a consultant.

camaira, 2020-09-30:
Hello,
I have the very same problem with the Galaxy A8 (A530F), M31 and A21. I suspect it has something to do with the bluetooth stack.
Do yo have any further infos on this topic? Any help is highly appreciated. Thank you.