AltBeacon/android-beacon-library #948: [NoClassDefFoundError] Crash on start Ranging Beacons In Region

matdev, 2020-02-06:
### Expected behavior

Library starts ranging.

### Actual behavior

`java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/localbroadcastmanager/content/LocalBroadcastManager;`

```
org.altbeacon.beacon.BeaconLocalBroadcastProcessor.unregister + 80 (BeaconLocalBroadcastProcessor.java:80)
--
  | org.altbeacon.beacon.BeaconLocalBroadcastProcessor.register + 72 (BeaconLocalBroadcastProcessor.java:72)
  | org.altbeacon.beacon.service.ScanJobScheduler.ensureNotificationProcessorSetup + 70 (ScanJobScheduler.java:70)
  | org.altbeacon.beacon.service.ScanJobScheduler.schedule + 130 (ScanJobScheduler.java:130)
  | org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob + 85 (ScanJobScheduler.java:85)
  | org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob + 92 (ScanJobScheduler.java:92)
  | org.altbeacon.beacon.BeaconManager.applyChangesToServices + 1003 (BeaconManager.java:1003)
  | org.altbeacon.beacon.BeaconManager.startMonitoringBeaconsInRegion + 938 (BeaconManager.java:938)
  | org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.onBeaconServiceConnect + 203 (RegionBootstrap.java:203)
  | org.altbeacon.beacon.BeaconManager.bind + 420 (BeaconManager.java:420)
  | org.altbeacon.beacon.startup.RegionBootstrap.<init> + 105 (RegionBootstrap.java:105)
```

### Steps to reproduce this behavior
Replaced the 2.16.3 lib with new version 2.16.4 

> implementation 'org.altbeacon:android-beacon-library:2.16.3'

Build a release APK
Run the APK and start scanning for BLE devices and beacons

**Crash is not reproduced when running the app locally from Android Studio**

### Mobile device model and OS version

OnePlus A6003 Android 10

### Android Beacon Library version

 2.16.4

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-02-06:
Do you have proguard enabled?  If you do, does the problem go away if you disable it and do a release build?

matdev, 2020-02-10:
I do have the same problem after disabling proguard

davidgyoung, 2020-02-10:
Another report of this here:  https://stackoverflow.com/questions/60140990/getting-fatal-crash-using-startmonitoringbeaconsinregion

This certainly seems to be a real issue, but there must be another variable that we can't pinpoint.  I cannot reproduce this with the reference app with the latest library version if I build a release APK and install it on my phone.  

@matdev, can you please try an unmodified copy of the reference app https://github.com/altBeacon/android-beacon-library-reference, make a release build and then install and see if you get this crash?

If you  do not get a crash, then we can start looking for difference in the gradle files between your project and the reference app to look for a difference that triggers it.

matdev, 2020-02-11:
Thanks for the link to the stackoverflow question. Using the accepted answer, my no longer crashes using version 2.16.4 of the lib :) 

I've had to add the following code in my app/build.gradle though:

```
android.packagingOptions { 
    exclude 'META-INF/androidx.localbroadcastmanager_localbroadcastmanager.version' 
}
```

I'm unable to install the unsigned APK of the reference app. If you still want me to test it, can you please send me a signed APK ?

davidgyoung, 2020-02-28:
At least the StackOverflow post above provides a workaround.

closed 2020-02-28 by davidgyoung