AltBeacon/android-beacon-library #949: detection of beacon RSSI values continuously at 10Hz

Xiryl, 2020-02-20:
I created an application to detect beacons via the bluetooth module.

The Beacons are configured with Eddystone UID, protocol with an advertising interval of `100ms` and transmit power level of strong (`10dBm`).
What I would like to be able to do is to detect the `RSSI` value of the beacons with a frequency of `10Hz` (i.e. 100ms).

I've already prepared a service that implements the altbeacon library, the relevant part are showed below:

```java
mBeaconManager = BeaconManager.getInstanceForApplication(this);
        
        beaconRegion = new Region("beacon_region", null, null, null);

        // use Eddystone protocol
        mBeaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));

        mBeaconManager.setAndroidLScanningDisabled(true);
        mBeaconManager.setForegroundScanPeriod(100l); // scan frequency
        mBeaconManager.setForegroundBetweenScanPeriod(0);
```

and the callback `didRangeBeaconsInRegion` from which I see the beacons and get the RSSI value is this:

```java
@Override
    public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
        Log.i(BeaconDataCollectorService.class.getSimpleName(), String.format("Found %s beacons in range", beacons.size()));
    }
```

> What's my problem?
Running the app, I notice that within didRangeBeaconsInRegion, I often don't see all the beacons.

I ran the data collection for a few seconds, creating a csv format of the data obtained, where you can see that I often have 0s (indicating that the beacon was not detected); I put the sharing of this csv on [pastebin][2] (https://pastebin.com/zkUZC5R4)

How can I improve the scan frequency by always being able to detect all the beacons?

altbeacon version used:
```java
implementation 'org.altbeacon:android-beacon-library:2.16.3'
```

Android version: 9

Thanks

  [1]: https://github.com/AltBeacon/android-beacon-library
  [2]: https://pastebin.com/zkUZC5R4

davidgyoung, 2020-02-20:
Unfortunately, the library is not designed to do this, as the ranging APIs are modeled after iOS equivalents which give aggregate detections at fixed intervals as opposed to access to individual packets when they arrive.

The main reason you often get 0 beacons detected with such a short scan period is because the scan is turned on and off every 100ms, and it takes 10ms or so to start and stop.  This gives a good probability of missing the detection.

Here's the good news: If the library ever detects two beacon packets in the same scan period then it will stop turning the scan off and your detection rate will improve.  But getting it to detect two beacons in 100ms is nearly impossible of your beacon advertises at a nominal 10Hz.

One thing you might try is to start by setting the scan period to 1 second.  After you have detected a finite number of your beacons for 10 secs or so, there is a very good chance you will have detected two beacons in one scan cycle and then you can switch to a scan period of 100ms and get a higher detection rate.

You will never get a 100% detection rate for two reasons:

1. Not all beacon packets transmitted are received due to collisions and radio noise.  Al close range 80-90% is typical.
2. BLE advertisers randomize when their packets are sent to avoid collisions.  A nominal 10H transmitter might have anywhere between 70ms and 140ms between individual packets, so for fixed scan cycles of 100ms sometimes there will be 0, 1 or 2 Packers eligible to be received.

If you really need to get callbacks on every detection, you might try setting up a NonBeaconBleScanCallback in the BeaconManager, then calling BeaconManager.clearBeaconParsers().  This will cause all beacons detections to be sent to that callback immediately when detected.  You will then have to construct a new BeaconParser for use inside that callback to decide the beacon from the raw packet.  Use BeaconParser.parse(...)

davidgyoung, 2020-02-20:
Discussion moved here: https://stackoverflow.com/questions/60317899/altbeacon-detection-of-beacon-rssi-values-continuously-at-10hz/60319392#60319392

closed 2020-02-20 by davidgyoung