AltBeacon/android-beacon-library #952: Quick battery draining in Pixels, OnePlus devices

kalkanijayesh, 2020-02-26:
### Expected behavior
Everything is working fine except receiving battery-draining reports from the Pixels, OnePlus users. I have also added **BackgroundPowerSaver** class to minimize the battery draining issue.
`backgroundPowerSaver = new BackgroundPowerSaver(this);`
`beaconManager.setAndroidLScanningDisabled(true);`

FYI: I haven't modified any scanning time interval.
It should minimize the battery draining issue as like other devices.

### Actual behavior
AltBeacon library performs well in all the devices except Pixels, OnePlus. When iBeacon proximity status is FAR it consumes more battery as compared to NEAR/INTERMEDIATE status.

I have collected some data to get the exact idea about the battery draining issue.
You can see in the picture "Overnight Battery Discharge Rate" dramatically increasing while AltBeacon scanning ON.

[image]

### Steps to reproduce this behavior
1. Start any number of beacon monitoring with major-minor.
2. Charge your device battery with 100%
3. Kill your app -> Unplug your device -> Turn of the screen.
4. Keep as it is for a while (Min 2 hours, it would be great for 5-6 hours)
5. Check battery percentage and the time difference.

### Mobile device model and OS version
OS version: Android 8 or later
Mobile device: Google Pixels, OnePlus

### Android Beacon Library version
implementation 'org.altbeacon:android-beacon-library:2.16+'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-02-26:
Can you reproduce this problem with the reference app?

Does your app use a foreground service?

Exactly how are you measuring deep sleep time?

Does your app set beaconManager.setEnableScheduledScanJobs(false)?

How many beacons are visible to the phone during this test?  Are they consistently visible or not visible? (Detection timestamps would be helpful, either ranging detect times, or region enter/exit times.)

This report surprises me.  Under default settings, the library will use a scheduled job to do scans for 30 seconds every 15 minutes (and this should be slowed down by the OS to less frequently once the app enters deep doze for being motionless with the screen off).  Nothing should block deep doze mode and the phone should enter it in a few minutes under the conditions you describe.

If it is not going into Doze, something is wrong.

kalkanijayesh, 2020-02-27:
Q. Can you reproduce this problem with the reference app?
A. No, I haven't checked with the reference app. 

Q. Does your app use a foreground service?
A. No

Q. Exactly how are you measuring deep sleep time?
A. I think the device goes automatically into the deep sleep mode if no longer uses. 

Q. Does your app set beaconManager.setEnableScheduledScanJobs(false)?
A. No, I haven't used it.

Q. How many beacons are visible to the phone during this test? Are they consistently visible or not visible?
A. It might be 2-5 beacons.

FYI: I have turned off battery optimization while testing. Maybe this option prevents going into the Doze mode.

davidgyoung, 2020-02-27:
Yes, disabling battery optimizations will prevent the device from going into Doze mode.  In newer Android versions there are two variants of Doze:

Light Doze - entered shortly after the screen goes off
Deep Doze - entered when the phone is not in motion for a few minutes and the screen is off.

I have tested the reference application in the above scenarios with Android 10.  If you do not turn off battery optimizations and leave it on your nightstand overnight unplugged, it will go into Deep Doze quickly and no beacons will be detected.  Battery drain will be very very low.

If you disable battery optimizations, then it won't go into either Light Doze or Deep Doze.  This is really an unrealistic test scenario.   The phone will behave (in terms of battery) like it would with the screen on, albeit without the battery drain of illuminating the screen itself.

I'd recommend you redo your tests another way.  Perhaps:

1. Don't disable battery optimizations.  Leave the screen off and the test device in your pocket all day long -- or do something else to keep the device in periodic motion to keep it from staying in Deep Doze.
2. Use ADB to command it into light doze (and somehow block it from going into deep doze).  I'm not sure that is possible but you can investigate.  For this to work you'd have to set up ADB over WiFi so your ADB connection is not over USB keeping the device charging.

I know the above is tricky.  I've done lots of battery testing with this library, so I have faced these testing problems before.  ADB over WiFi is worth setting up to get these tests done the best way.

kalkanijayesh, 2020-02-28:
Let me try to collect the information accordingly. And let's see what happens. 

**FYI:** Sorry, I forgot to mention, I have used "[AccuBattery Pro](https://play.google.com/store/apps/details?id=com.digibites.accubattery)" to get the details about the Deep Sleep. 

That's why we are recommending to all our app users, please disable the battery optimization feature to detect the beacon in every device status. 

What should I need to do to detect the beacon in any status of the device ??? According to our app aim, we need accurate beacon detection along with minimal battery usage.

davidgyoung, 2020-02-28:
Well, @kalkanijayesh , if you are telling users to disable battery optimizations, then they should not be surprised if they get battery drain!  And, yes, the specific impact may be different on OnePlus and Pixel devices.

Given this critical piece of information, I am closing this issue.  Disabling battery optimizations is obviously the cause of the battery drain, so this is not a problem with this library.

Battery drain vs. rapid reliable beacon detection is a tradeoff.  You cannot have both.  If you decide you are willing to sacrifice battery to get rapid reliable detections then you must do two things:

1. Use a Foreground Service (see instructions here: https://altbeacon.github.io/android-beacon-library/foreground-service.html)
2. Consider obtaining a wake lock for your app to prevent the phone from going into deep doze.  If you do not obtain a wake lock and the phone goes into deep doze, you will not get detections.

The above two points are the most extreme things you can do to get constant detections reliably, but the battery impact is significant and noticeable.   There are lesser things you can do with less battery impact, but the detection reliability and speed will be impacted negatively.  Again, it is a tradeoff.

If you need assistance finding the right tradeoff for your app, I am happy to help further in a forum appropriate for a custom app.  You can post questions on  StackOverflow.com, or contact me directly through my github profile if you need dedicated consulting assistance.

closed 2020-02-28 by davidgyoung