AltBeacon/android-beacon-library #953: Unexpected bluetooth state changes on Samsung S9

Maartinsh, 2020-03-10:
### Expected behavior
Do no disable Bluetooth **always**

### Actual behavior
Whenever Samsung S9 (I can reproduce issue only on this device right now) has made a connection (for instance with headphones), bluetooth scanning [AltBeacon library] will cause losing already made BLE connection to headphones.

### Steps to reproduce this behavior
Connect to any BLE device and start beacon scanning

### Mobile device model and OS version
Galaxy S9, Android 10

### Android Beacon Library version
2.16.4

### Logs

```
D/CycledLeScanner: cancel wakeup alarm: null
D/CycledLeScanner: Set a wakeup alarm to go off in 9223372035824084041 ms: PendingIntent{feef327: android.os.BinderProxy@dfd99d4}
I/BeaconService: start ranging received
D/BeaconService: Currently ranging 1 regions.
D/CycledLeScanner: start called
D/CycledLeScanner: starting a new scan cycle
D/CycledLeScanner: starting a new bluetooth le scan
D/CycledLeScannerForLollipop: starting a scan in SCAN_MODE_LOW_LATENCY
D/CycledLeScannerForLollipop: Using a wildcard scan filter on Samsung because the screen is on.  We will switch to a non-empty filter if the screen goes off
D/CycledLeScannerForLollipop: registering SamsungScreenOffReceiver org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$5@4cd9ac3
D/CycledLeScannerForLollipop: Making new Android L scanner
D/BluetoothAdapter: STATE_ON
D/CycledLeScanner: Waiting to stop scan cycle for another 3000 milliseconds
D/CycledLeScanner: Scan started
D/BluetoothAdapter: STATE_ON
D/CycledLeScanner: Set scan periods called with 3000, 5000 Background mode must have changed.
D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
D/BluetoothAdapter: STATE_ON
D/CycledLeScanner: cancel wakeup alarm: PendingIntent{feef327: android.os.BinderProxy@dfd99d4}
D/CycledLeScanner: Set a wakeup alarm to go off in 9223372035824084012 ms: PendingIntent{feef327: android.os.BinderProxy@dfd99d4}
D/BluetoothAdapter: BLE support array set: 010011
D/BluetoothLeScanner: Start Scan with callback
D/BluetoothLeScanner: onScannerRegistered() - status=133 scannerId=0 mScannerId=0
D/BluetoothLeScanner: Registration failed, unregister scannerId = 0
D/BluetoothLeScanner: Scan failed, reason: app registration failed
E/CycledLeScannerForLollipop: Scan failed: app cannot be registered
D/BluetoothMedic: Broadcast notification received.
D/BluetoothMedic: Detected a SCAN_FAILED_APPLICATION_REGISTRATION_FAILED.  We need to cycle bluetooth to recover
D/BluetoothMedic: Power cycling bluetooth
D/BluetoothMedic: Power cycling bluetooth
D/BluetoothMedic: Turning Bluetooth off.
D/BluetoothAdapter: disable()
I/TestBeaconManager$BluetoothBroadcastReceiver: Turning Bluetooth off...
```

I understand that _SCAN_FAILED_APPLICATION_REGISTRATION_FAILED_ is related with bluetooth stack and generally, there's not a lot to do, but solution to disable/enable bluetooth connection makes very bad user experience (losing previous connections). Also, this does not happen rarely. I can reproduce the issue 10 times out of 10. Any ideas how to solve this?

davidgyoung, 2020-03-10:
You can certainly disable automatic bluetooth power cycling to correct errors by not enabling it with the BluetoothMedic (this is not automatic so you must have it configured in your app -- search for BluetoothMedic in your source code if you don't remember setting this up.)  Of course, not doing that may leave the bluetooth stack in a hung state if it gets that way.

In the log shown above, the scan error seems to happen when switching from background to foreground.  It doesn't look like the existing scan is stopped before a new scan is started.  I wonder if the exiting scan were properly stopped if it would have avoided the error you see.

Maartinsh, 2020-03-10:
> stopped if it would have avoided th

Thanks for response. I'll look into configuration, thanks for pointing that out. 

>  It doesn't look like the existing scan is stopped before a new scan is started. I wonder if the exiting scan were properly stopped if it would have avoided the error you see.

There was no existing scanning before. Scanning was started manually right after user brought application to foreground and clicked button.

> You can certainly disable automatic bluetooth power cycling to correct errors by not enabling it with the BluetoothMedic

What are your suggestions? Should it be by default on? Would it be wise to set this configuration based on the BLE connected devices - for instance, if there's at least single device connected, set it to _false_ (to avoid bad UI), else set it to _true_?

closed 2020-03-11 by Maartinsh