AltBeacon/android-beacon-library #955: I can't detect beacons in the  kindelf device is not pair No beacons detected

PongPloy2016, 2020-03-17:
kindelf  is mean device id for Intelligent anti-lost: When the connection is disconnected, the phone and device will alarm at the same time to ensure that the belongings will not be lost. 

kindelf is play store : https://play.google.com/store/apps/details?id=com.lenzetech.kindelf&hl=th 

and when used connected beacon other but  can be used normally 
[image]

but when entering this library to use And noticed that. this No beacons detected 

[image]

i set up code this : 

//setUpBeacon 

`    open fun setUpBeacon(applicationContext: Application) {
        setUpbindServise()

        backgroundPowerSaver = BackgroundPowerSaver(applicationContext);
        mBeaconManager?.getBeaconParsers()
            ?.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"))

        val identifiers = ArrayList<Identifier>()
        mRegion = Region(ALL_BEACONS_REGION, identifiers)

//         val region = Region("backgroundRegion", null, null, null)
//          regionBootstrap = RegionBootstrap(this, region)
        mBeaconManager?.setForegroundScanPeriod(DEFAULT_SCAN_PERIOD_MS)
//        กำหนดระยะเวลาการสแกนเป็น 1.1 วินาที
        mBeaconManager?.backgroundScanPeriod = 1100L
        // set the time between each scan to be 1 hour (3600 seconds)
        //We recommend not reducing the scan period to be less than 1.1 seconds  https://altbeacon.github.io/android-beacon-library/foreground-service.html
        mBeaconManager?.foregroundScanPeriod = 1100L
//        ตั้งเวลาระหว่างการสแกนแต่ละครั้งเป็น x ชั่วโมง (xxxx วินาที)
        mBeaconManager?.backgroundBetweenScanPeriod = 0
        mBeaconManager?.foregroundBetweenScanPeriod = 1100L
        // Enlazar al servicio de beacons. Oฃbtiene un callback cuando esté listo para ser usado
        mBeaconManager?.bind(this)
    }
`

//onBeaconServiceConnect

` override fun onBeaconServiceConnect() {
        Log.e(TAG, "onBeaconServiceConnect")

        if (Build.VERSION.SDK_INT == Build.VERSION_CODES.P) {
            try {
                // mBeaconManager = BeaconManager.getInstanceForApplication(context!!);
                checkNotiSound = true
                mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)

                //  mContext!!.toast(getString(R.string.start_looking_for_beacons))

            } catch (e: RemoteException) {
                //  context!!.toast(e.message!!)
            }

            mBeaconManager!!.addRangeNotifier(this)

            //TODO :: check toast anorid p
            //  Toast.makeText(applicationContext, "android P" + "\t" , Toast.LENGTH_LONG).show()

        } else {

            //android  7 , 7.1  to Test device

            mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)

            mBeaconManager?.addRangeNotifier(RangeNotifier { beacons, region ->
                Log.d("beaconManager", "didRangeBeaconsInRegion service" + beacons.size)
                Log.e("beaconManager prompt", "didRangeBeaconsInRegion $beacons")

                //TODO :: check toast anorid 7

                // Toast.makeText(applicationContext, "android 7" + "\t" + beacons.toString(), Toast.LENGTH_LONG).show()
                getBeaconMutableCollection(beacons)

            })

            checkNotiSound = true

            startRegionSet()

        }

    }
`

 override fun onBeaconServiceConnect() {
        Log.e(TAG, "onBeaconServiceConnect")

        if (Build.VERSION.SDK_INT == Build.VERSION_CODES.P) {
            try {
                // mBeaconManager = BeaconManager.getInstanceForApplication(context!!);
                checkNotiSound = true
                mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)

                //  mContext!!.toast(getString(R.string.start_looking_for_beacons))

            } catch (e: RemoteException) {
                //  context!!.toast(e.message!!)
            }

            mBeaconManager!!.addRangeNotifier(this)

            //TODO :: check toast anorid p
            //  Toast.makeText(applicationContext, "android P" + "\t" , Toast.LENGTH_LONG).show()

        } else {

            //android  7 , 7.1  to Test device

            mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)

            mBeaconManager?.addRangeNotifier(RangeNotifier { beacons, region ->
                Log.d("beaconManager", "didRangeBeaconsInRegion service" + beacons.size)
                Log.e("beaconManager prompt", "didRangeBeaconsInRegion $beacons")

                //TODO :: check toast anorid 7

                // Toast.makeText(applicationContext, "android 7" + "\t" + beacons.toString(), Toast.LENGTH_LONG).show()
                getBeaconMutableCollection(beacons)

            })

            checkNotiSound = true

            startRegionSet()

        }

    }

    fun stopBeacon() {

        // mBeaconManager = BeaconManager.getInstanceForApplication(this)

        this?.let {

            // BeaconManager.getInstanceForApplication(ContextSingleton.instance?.context!!).unbind(this)
            // mBeaconManager = BeaconManager.getInstanceForApplication(ContextSingleton.instance!!.context);
            mBeaconManager?.stopRangingBeaconsInRegion(mRegion!!)
            mBeaconManager?.stopMonitoringBeaconsInRegion(mRegion!!)
        }
            .let {
            }

    }

    fun startRegionSet() {
        try {
            if (!startRegion) {
                startRegion = true
                stopRegion = false
                beaconsSeen = HashSet<Beacon>()

                val identifiers = ArrayList<Identifier>()
                mRegion = Region(ALL_BEACONS_REGION, identifiers)
                //  mBeaconManager = BeaconManager.getInstanceForApplication(context!!)
                mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)
                mBeaconManager?.startMonitoringBeaconsInRegion(mRegion!!)
                //  mBeaconManager?.startRangingBeaconsInRegion(Region("myRangingUniqueId", null, null, null))
                //mBeaconManager?.startMonitoringBeaconsInRegion(Region("myMonitoringUniqueId", null, null, null))
                Log.d("beaconManager", "startRegion")
            }
        } catch (e: RemoteException) {
            e.printStackTrace()
        }

    }

    override fun unbindService(serviceConnection: ServiceConnection) {
        ContextSingleton.instance?.context?.unbindService(serviceConnection)
    }

    override fun bindService(
        intent: Intent?, serviceConnection: ServiceConnection, i: Int
    ): Boolean {
        return ContextSingleton.instance?.context?.bindService(intent, serviceConnection, i)!!
        // return false
    }

`    fun stopBeacon() {

        // mBeaconManager = BeaconManager.getInstanceForApplication(this)

        this?.let {

            // BeaconManager.getInstanceForApplication(ContextSingleton.instance?.context!!).unbind(this)
            // mBeaconManager = BeaconManager.getInstanceForApplication(ContextSingleton.instance!!.context);
            mBeaconManager?.stopRangingBeaconsInRegion(mRegion!!)
            mBeaconManager?.stopMonitoringBeaconsInRegion(mRegion!!)
        }
            .let {
            }

    }

    fun startRegionSet() {
        try {
            if (!startRegion) {
                startRegion = true
                stopRegion = false
                beaconsSeen = HashSet<Beacon>()

                val identifiers = ArrayList<Identifier>()
                mRegion = Region(ALL_BEACONS_REGION, identifiers)
                //  mBeaconManager = BeaconManager.getInstanceForApplication(context!!)
                mBeaconManager!!.startRangingBeaconsInRegion(mRegion!!)
                mBeaconManager?.startMonitoringBeaconsInRegion(mRegion!!)
                //  mBeaconManager?.startRangingBeaconsInRegion(Region("myRangingUniqueId", null, null, null))
                //mBeaconManager?.startMonitoringBeaconsInRegion(Region("myMonitoringUniqueId", null, null, null))
                Log.d("beaconManager", "startRegion")
            }
        } catch (e: RemoteException) {
            e.printStackTrace()
        }

    }

    override fun unbindService(serviceConnection: ServiceConnection) {
        ContextSingleton.instance?.context?.unbindService(serviceConnection)
    }

    override fun bindService(
        intent: Intent?, serviceConnection: ServiceConnection, i: Int
    ): Boolean {
        return ContextSingleton.instance?.context?.bindService(intent, serviceConnection, i)!!
        // return false
    }

`
this is. detail doc : 

[image]

Thank you

davidgyoung, 2020-03-17:
Assisting in setting up a project to use a specific manufacturer's beacon is beyond the scope of this form.   (This forum is reserved for bug reports and feature requests.)

I suggest you re-format this question to meet the standards of StackOverflow.com and post the question there.  I will be glad to assist in that forum.  Please include in your question:

* Have you tried an off-the-shelf beacon locator app for Android like BeaconScope?  Does that app detect this device?
* Have you tried to use a beacon simulator app on a different phone (e.g. BeaconScope for Android or Locate Beacon for iOS) to see if your program can detect that beacon?

closed 2020-03-17 by davidgyoung

PongPloy2016, 2020-03-18:
OK  bro Thank you to Answer  suggest  ^__^

I copy source code and try use  ' '  insert code but it can not reformate as standard code and I think used  kindelf device, not beacons because it is  Bluetooth on deviceV.4.0. As announced in the document sheet I research used app other together with found  kindelf device (in photo)

[image]

this name is "iTag" is Protocal : Low Energy