AltBeacon/android-beacon-library #957: didRangeBeaconsInRegion Called Several Times for same Beacon.

mathieu56000, 2020-04-02:
### Expected behavior
During a beacon scan, when a beacon is found, we want to enter in the "didRangeBeaconsInRegion" function just once.

### Actual behavior
Each time, instead of just one entering in the "didRangeBeaconsInRegion" function, we observe that when a beacon is detected, in the logs, we have 2 lines for the same beacon. The RSSI is always the same (for the repetition) and between the 2 logs, we have a time between 10ms and 40ms.
[image]

### Steps to reproduce this behavior
Here is the Periods that we have programmed :
```
beaconManager.setEnableScheduledScanJobs(false);
beaconManager.setBackgroundBetweenScanPeriod(1000);
beaconManager.setBackgroundScanPeriod(1000);
beaconManager.setForegroundBetweenScanPeriod(1000);
beaconManager.setForegroundScanPeriod(1000);
```

And here is the "didRangeBeaconsInRegion" function. The problem is that the function "BLELaunch()" is called twice (or more) each time and we just want to call the function once. beacons.size() is always 1 by the way.
```
@Override
    public void onBeaconServiceConnect() {
        RangeNotifier rangeNotifier = new RangeNotifier() {
           @Override
           public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
              if (beacons.size() > 0) {
                  for (Beacon beacon: beacons) {
                      if (beacon.getRssi() > -60.0) {
                          Log.d(TAG, "Beacon dans Region : " + beacon.getRssi());
                          BLELaunch();
                          //scanLeDevice(true);
                      }

                  }
              }
           }

        };
```

### Mobile device model and OS version
Oneplus6 , Android 10 
OxygenOs V10.3.1

### Android Beacon Library version
implementation 'org.altbeacon:android-beacon-library:2.16.4'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-04-02:
Is it possible the code has called beaconManager.startRangingBeaconsInRegion(...) twice, using two different regions that both match the beacon?  Can you please show your code that starts ranging?

mathieu56000, 2020-04-02:
Thank you for your quick answer!
We have just one "startRangingBeaconsInRegion" in our code.
We notice that it happens too with the reference app available on GitHub. When we "Disable Monitoring" and when we "Start Ranging" again, we have one more line in the "didRangeBeaconsInRegion" function.
Example, after 3 switch, we observe that in the logs :
[image]
We can see that there are blocks of 3 lines each times...

We saw on another issues in that GitHub that there was the same issue when bind and unbind several times. In the issue #735  , the problem seems to be solve but it doesn't work for us...
Have you some ideas for helping us ?

Thank you,

mathieu56000, 2020-04-02:
Hi David,
We finally managed to fix the problem thanks to the issue #810. Thank you for the help and have a good day!

closed 2020-04-03 by mathieu56000