AltBeacon/android-beacon-library #959: Retrieve real calibrated tx power when transmitting beacon signal

michaellee8, 2020-04-03:
### Expected behavior
The calibrated tx power is advertised when the device is advertising beacon signal

### Actual behavior
Arbitrary tx power value is sent.

### Steps to reproduce this behavior
Follow the instructions in https://altbeacon.github.io/android-beacon-library/beacon-transmitter.html

### Mobile device model and OS version
Samsung S10

### Android Beacon Library version
2.16.4

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-04-03:
Can you please show a simple test case to demonstrate this that includes:

1. Code you used to transmit setting the txPower to a specific value
2. Code used to scan that shows a different txPower value

I think the most likely explanation is some issue with the testing procedure -- this functionality is pretty fundamental and to my knowledge has been working fine for 5 years.   But either way, I believe providing the information above will shed light on the issue you are seeing.

michaellee8, 2020-04-04:
@davidgyoung Currently, the library requires the developer to set the tx power value to a hardcoded one:
```
Beacon beacon = new Beacon.Builder()
       	.setId1("2f234454-cf6d-4a0f-adf2-f4911ba9ffa6")
       	.setId2("1")
       	.setId3("2")
        .setManufacturer(0x0118) // Radius Networks.  Change this for other beacon layouts
        .setTxPower(-59)
        .setDataFields(Arrays.asList(new Long[] {0l})) // Remove this for beacon layouts without d: fields
        .build();
// Change the layout below for other beacon types
BeaconParser beaconParser = new BeaconParser()
        .setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25");
BeaconTransmitter beaconTransmitter = new BeaconTransmitter(getApplicationContext(), beaconParser);
beaconTransmitter.startAdvertising(beacon, new AdvertiseCallback() {

            @Override
            public void onStartFailure(int errorCode) {
                Log.e(TAG, "Advertisement start failed with code: "+errorCode);
            }

            @Override
            public void onStartSuccess(AdvertiseSettings settingsInEffect) {
                Log.i(TAG, "Advertisement start succeeded.");
            }
        });
```

However, such hardcoded value is unlikely to reflect the actual power as emitted by the android device's signal. How do I retrieve the actual power transmitted by the device? Are there any API to retrieve it?

davidgyoung, 2020-04-04:
No, it is not possible to use any APIs to determine the proper "tx power" value (also known as "measured power" or "calibrated power" for a beacon.  You must do a calibration procedure to measure this in real-world conditions.

> A beacon transmission includes a transmitter power field that indicates how strong the signal should be at a known distance. As an example, when using iBeacon technology, the standard measurement is the signal level that an iPhone 5s (the latest model available when iOS 7 was released) will measure for a beacon at a distance of one meter. Beacons must be calibrated by measuring the signal level (known as Received Signal Strength Indicator or RSSI) at this reference distance and then configuring the beacon to transmit reference value. Values of RSSI are measured in dBm, and a typical one meter calibration value for a beacon transmitter will be -59 dBm. The actual calibration value for different beacons varies because each may have a transmitter and antenna that puts out a different amount of radio power. Some beacons, such as Radius Networks’ RadBeacon products, have configurable transmitter output power.

Read more [here](http://www.davidgyoungtech.com/2014/12/04/fundamentals-of-beacon-ranging)

closed 2020-04-04 by davidgyoung