AltBeacon/android-beacon-library #961: Foreground Service Transmitter Enchantment (Corna Virus open app)

thorgod, 2020-04-09:
Working on an open source social distancing app for the Corna Virus. Pretty straight forward, if within 6 feet of someone with the same app it will beep. (At my work people are not social distancing)

Currently I know there is no code or examples out there for Foreground service for transmitting. My knowledge if fairly new, so adding transferring the service you already have for receive is a bit daunting to me.

I don't expect this enchantment to get reject because I saw an older post with the same request. 
(https://github.com/AltBeacon/android-beacon-library/issues/291)

It might be to hard to do the transmitting and receiving in the same service but its worth a try to get it out there.

davidgyoung, 2020-04-09:
@thorgod, the library already supports using a foreground service to keep a transmitter going in the background.  I have added a coding example to the documentation [here](https://altbeacon.github.io/android-beacon-library/beacon-transmitter.html)

closed 2020-04-09 by davidgyoung

thorgod, 2020-04-09:
Thanks for the examples, going to test it out now. =) I appreciate the help