AltBeacon/android-beacon-library #963: Crash in ScanJob.java line 145 onStartJob()

matdev, 2020-04-14:
### Expected behavior
No crash

### Actual behavior
Crashes are seen occuring in our app's Crashlytics console
Surprisingly, there is no stack trace available in Crashlytics

### Steps to reproduce this behavior

### Mobile device model and OS version
Android 8 (BQ, LGE)

### Android Beacon Library version
2.16.4

davidgyoung, 2020-04-14:
Unfortunately, without additional information including an Exception type, a proper line number and a stack trace, there is nothing that can be done to track this down.   Line number 145 of ScanJob.java is whitespace in the source file for 2.16.4, so even that single detail is suspect.   

If you can gather more information, please open a new issue.  But since there is insufficient information to investigate, there is no point in leaving an unresolvable issue open.

closed 2020-04-14 by davidgyoung

matdev, 2020-04-15:
I understand, I'll you know if I manage to get the stacktrace.
For info, all of these crashes occured when app is in background