AltBeacon/android-beacon-library #964: Beacon not detect in samsung

dhavalrshingala, 2020-04-14:
### Expected behavior
When i am scan beacon it not detect from ios device and some of latest android devices most of samsung

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version
Samsung note 9 with Android 10 OS and Iphone latest Os

### Android Beacon Library version
2.16.4
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-04-14:
Cases where beacons cannot be detected are almost always due to programming issues in a custom app or problems with the beacon used to test that app.  

Please go through the suggested steps in the answer to this question:  https://stackoverflow.com/a/60915235/1461050

If you still have trouble, please verify you have the same problem with the reference app [here](https://github.com/altBeacon/android-beacon-library-reference).  If you can reproduce the problem with the reference app, please open a new question, providing the details from the "things to check" section of the post on StackOverflow.com

closed 2020-04-14 by davidgyoung

dhavalrshingala, 2020-04-20:
In lower version work perfect only not detect in android latest OS i have Samsung note 9 with latest Android OS.

davidgyoung, 2020-04-21:
@dhavalrshingala  please see troubleshooting steps here: https://altbeacon.github.io/android-beacon-library/detection-trouble.html