AltBeacon/android-beacon-library #967: Where to find Updated android-distance.json?

gbounch, 2020-04-25:
Looking for android-distance database , starting  from the website the refering url is broken
https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/resources/model-distance-calculations.json

I have found a file in source code but has only 3 models.
Is there a more general database having more models somewhere?

Thank You

davidgyoung, 2020-04-25:
Thanks for the report of the broken link @gbounch.   If you tell me where you found it, I will fix it.

There are two locations where this file is published:

1. Source Code.  This is the file bundled with the library, and which is used if no update is downloaded:  https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/resources/model-distance-calculations.json

2. Web update URL.  This is the location where a library that happens to have internet access will try and download an update on first launch:  https://s3.amazonaws.com/android-beacon-library/android-distance.json

I'm not sure this will help you though -- both files have just four entries.  It is fair to say that my plans for crowdsourcing the information that goes into these files has failed.  There are many reasons for that, chief among them the procedure for taking device-specific measurements and submitting them is too difficult and complex.

There has been renewed interest in collecting similar data as a result of using this technology for novel coronavirus contact tracing apps.  So I would be very interested in revamping the way the library collects and stores these data and merging the information with other data sources.

I would love for others to work with me on this effort if you have time and interest or know those who do.

gbounch, 2020-04-25:
Thank you the broken link is in this page https://altbeacon.github.io/android-beacon-library/distance-calculations.html.
We are thinking about using this project to create a proximity tracking app to be usefull for enterprises (there are some features that an enterprise needs that the public app will not have) . So we can think about creating actions to help you creating this database. 
It would be helpfull to find a simple way only to trace the Droplet distance. I think that is different to have values for calculating every distance than have the values at only 2 meters. So i think that the process of taking device-specific measurements can be simpler. What do you think about this?
I think that defining a range of values that can be received near the 2 meters distance could be good . This range can depends on the devices but also on the environment condition : position in clothes, walls  and so on. So, do you think can exists an everage values of signal strenghts with a set of loss values in various conditions (a wall, in the jacket and so on ) could be more useful...

Probably i have to put this question in the documents/#188

gbounch, 2020-04-25:
...thinking more. If it is so difficult to create the database, is it  possible tu use the hype about Covid to create it with the users help?
I think the precision problem could be simpler to solve for the COVID use case.
I mean that it can be simpler to create a database  to define the signal strenght at the droplet distance (2 - 3 mt) for various devices models couples related to  environmtental conditions.
Basicaly we need to Know the everage segnal loss for enviromental spaces :  something like  0 db in open space , -xx db in closed spaces , - yy db in user jacket, -zzdb off the walls and so on... 
So simplyfied i can use the user feedback: 
the app allarms in proximity at droplet distance, if it is wrong user can help inputing the real distance, i can have the two models of the devices are connecting and i can store the signal strength. 
After having collected multiple data of the same model devices couples i can create an everage of the signals. Using the data about the ipotetical conditions of the environment i think we can create a procedures that can "argue" about the environmental conditions that make variations from the everage.
So we dont'have to create a database manually but it is created by the users....

davidgyoung, 2020-04-25:
Thanks for the thoughts @gbounch.  Some quick feedback:

1. I think it is important to separate three distinct problems:  (a) characterize and document device-specific receiver RSSI variations (b) characterize and document device-specific transmitter RSSI variations (c) characterize environmental effects of RSSI measurements (open space, closed space, jacket, etc.)  

2. There are two big obstacles to getting crowdsourced measurements of device-specific receiver RSSI variations (a) each user needs a reference bluetooth transmitter with a known signal power level.  (Without a known reference, it is impossible to measure device-specific reception variations.  (b) You must make a procedure that is easy for a layperson to follow to get accurate and reliable measurements.  I don't have a good solution for either of these obstacles.

3. In my opinion it will never be possible to build a system that automatically measures device specific measurements based on 1a, 1b and 1c.    There are just too many variables -- far more than have been even discussed here.  Only by breaking down the problem and solving its components can you hope to make progress.

gbounch, 2020-04-25:
Ok, but isn't thera a semplification for covid case: i need RSSI only at 2 meters.
 If i can create a huge database of signal received in real situations by all the device model couples in the market at droplet distance, is it not possible that the prediction  became more and more accurate? 

In our use case (in enterprise), i can have a limed number of interaction and can ask to users to tell me when my application is wrong about the allarm. i can ask to go far as 2 meters and i can record the signal recieved by the 2 devices.
The next time the 2 devices are near (and in enterprises it is possible that the enviromental conditions are similar) i can use stored data to correct my prediction.
In outside world the same can be acheived at device model level (not device itself) whit the variations of the environmental condition. For example : if i have a prediction that in everage the rssi at 2 meters is -65db an that the variations are from +10db and -10db i can leverage the allarm or mark the interaction as "in droplet range" if the signal is from 0 to -75 fb. 
I surely simplifying too much but we need some semplification to let this technology be helpful for the covid use case.

davidgyoung, 2020-04-26:
>> if I can create a huge database of signal received in real situations by all the device model couples in the market at droplet distance, is it not possible that the prediction became more and more accurate?

The problem here is that how do you know when the two devices are "at droplet distance"?  You need a second sensor to tell you when this is true so you can correlate the bluetooth measurement taken at that time.  I don't know of any such second sensor that can provide this measurement.

gbounch, 2020-04-26:
I am asking to the user! The app tells: "Go at 2m far away from the other user and press this huge button on the app". 
Italian Government has choosen an application that probably will have the same kind of problems.
If the idea of ​​having such a large database, involving users can be useful, then we can do 2 things:
1) Start from the companies and involve the employees in calibration activities (the application sounds the alarm, the user signals that he is distant so that the alarm is wrong, the application asks to approach 2 meters and press a big button red on the app)
2) Create events in schools (when they reopen) or in squares where people approach and press the big red button.

davidgyoung, 2020-04-26:
I think the real challenge here is that users typically won't be able to accurately estimate a 2 meter distance.

gbounch, 2020-04-27:
I made the same proposal here ...
https://github.com/DP-3T/documents/issues/188#issuecomment-619914240
let's see what they think

mellison1, 2020-04-27:
> Thanks for the report of the broken link @gbounch. If you tell me where you found it, I will fix it.
>
> There are two locations where this file is published:
>
> 1. Source Code.  This is the file bundled [...]

The web updated url is missing, causing an error on my side for ranging beacons.

`W/DistanceConfigFetcher: error reading beacon data
    java.io.FileNotFoundException: https://s3.amazonaws.com/android-beacon-library/android-distance.json
        at com.android.okhttp.internal.huc.HttpURLConnectionImpl.getInputStream(HttpURLConnectionImpl.java:251)
        at com.android.okhttp.internal.huc.DelegatingHttpsURLConnection.getInputStream(DelegatingHttpsURLConnection.java:210)
        at com.android.okhttp.internal.huc.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:26)
        at org.altbeacon.beacon.distance.DistanceConfigFetcher.request(DistanceConfigFetcher.java:94)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceUpdater.doInBackground(ModelSpecificDistanceUpdater.java:28)
        at org.altbeacon.beacon.distance.ModelSpecificDistanceUpdater.doInBackground(ModelSpecificDistanceUpdater.java:16)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
W/ModelSpecificDistanceCalculator: Cannot updated distance models from online database at java.io.FileNotFoundException: https://s3.amazonaws.com/android-beacon-library/android-distance.json
`

davidgyoung, 2020-04-27:
This was a file permissions issue on Amazon S3, @mellison1.  It is resolved now.

schauersbergern, 2020-05-18:
I have taken reference measurement with a Samsung A20 E, Huawei p30 pro, Huawei p30 lite, Samsung Galaxy S20, Samsung A71, Samsung A5 and Huawei Mate.

I want to share the values with your library but I have two issues 

- Your Reference Calculations are wrong and I don't know how to do it right.

When I enter the independent and dependent regression variables from here:
https://docs.google.com/spreadsheets/d/1ymREowDj40tYuA5CXd4IfC4WYPXxlx5hq1x8tQcWWCI/edit
and from here:
https://altbeacon.github.io/android-beacon-library/distance-calcs/nexus5.html

In the power regression calculators found here and here:
http://www.xuru.org/rt/powr.asp
https://keisan.casio.com/exec/system/14059931777261

I get the same values for A and B constants in the online tools but they differ completely from the values you have provided. What am I doing wrong here?

- How is the 1m reference measurement from the iphone added to the test scenario? Is this just detached from the test itself, or does the reference iPhone have to be the sender of the measured beacons?

davidgyoung, 2020-05-18:
@schauersbergern, apologies, it seems somebody else edited the spreadsheet (in 2017!) and modified a value in a single cell -- the reference power value in Table 1.  This affected all other cells (they were derived calculations), so this is what caused the regression calculations not to match.  This is why you were not able to reproduce the calculations.

I just reverted that change, and now the regressions are reproducible again.    

The reason there is a 1 meter distance measurement from an iPhone is because it is an independent measurement of the strength of the signal at one meter that is independent of the device under test.  Understand that every Android phone out there  has a bluetooth receiver that is more or less sensitive.  They are all over the map!

The known reference value was originally defined as an iPhone 4S, 5 or 5s, which all had very similar RSSI sensitivities.  But hardly anybody has one of those around anymore.  Assuming you don't have one of these, another iPhone is really the best you can do.  You just have to document which model you used for your measurement.

davidgyoung, 2020-05-19:
By the way, if you don't mind, please share your raw measurements, perhaps as Google Doc spreadsheets linked from a new Github issue.  Thank you for the help!

schauersbergern, 2020-05-22:
Hello @davidgyoung,
thanks for advice, I added a iPhone 8 reference measurement and opened following pull request:
https://github.com/AltBeacon/android-beacon-library/pull/972

The data is calculated from following rssi measurements:
https://docs.google.com/spreadsheets/d/185q5AJmWipH23si5IAvMpdy3rGyuJFsCwkqM6EkqFH8

The measurements where carried out at smartfactory owl, Germany, they also posted about on linkedin: https://www.linkedin.com/company/smartfactoryowl/

br, Nik