AltBeacon/android-beacon-library #978: Distance calculation Algorithm

shanmugpriyan306, 2020-06-04:
My questions are around the distance calculation data specified here: https://altbeacon.github.io/android-beacon-library/distance-calcs/nexus5.html

Is there any official documentation available for reason behind calculation of these distance data**[Intercept, Multiplier, Power]** Any algorithm. I would like understand more about. 

Why we need to use only iPhone 5s @1 meter RSSI value for Power Curve Distance formulat. What is logic behind it? Are we free to use any other mobile RSSI value @ 1 meter distance also.

davidgyoung, 2020-06-04:
The parameters shown are the the elements of a power curve when doing a regression.

A curve fitting technique with a power curve is used  because the equation offered by physics doesn’t work as well in practice because there are other factors going on that involve more than radio signal theory.

Every phone differs in the way it measures signal strength based on sensitivity and other factors.  This is why we must have a standard reference.  An iPhone 4s/5/5s/SE 2016 all have very similar RSSI measurements so can serve as good references.  Other phones can be used if they have known differences from these reference models.  Such conversions are shown here https://docs.google.com/spreadsheets/d/1-voNJAeHz78AarZmfds8WiplNhUDzeYrsW0YzKfrExU/edit?usp=drivesdk

Read more in the Fragmentation section of this blog post: 
http://www.davidgyoungtech.com/2020/05/15/how-far-can-you-go

shanmugpriyan306, 2020-06-08:
Thanks @davidgyoung 

I went through spreadsheet. I'm not clear on few things on it.
How to measure Transmitter Power **1 meter, 1 meter @ -90, 1 meter @ +90, 2 meter**  on my knowledge transmitter power cannot be positive i.e (+90). Highlighted my question below on image.
[image]

How to measure Receiver sensitivity **1 meter, 1 meter @ -90, 1 meter @ +90, 2 meter, 5 meter** on my knowledge transmitter power cannot be positive i.e (+90). Highlighted my question below on image.
[image]

davidgyoung, 2020-06-09:
For these measurements, I placed the transmitter on top of a plastic camera tripod elevated 1.5 meters above the ground.  The transmitter's primary axis (0 degrees)  is defined as the opposite side of the USB connector for a USB beacon or the screen for a phone beacon. 

The "1 meter"  measurement was taken with the phone at the same vertical elevation as the trasmitter, with the back of the phone receiver aimed at the beacon along the primary axis.

The "1 meter at -90" was taken 90 degrees off of the primary axis.  In other words, the back of the phone receiver is still aimed at the beacon transmitter, but at a position on the right-hand side of the transmitter.

The "1 meter at +90" was taken 90 degrees off of the primary axis in the other direction.  In other words, the back of the phone receiver is still aimed at the beacon transmitter, but at a position on the left-hand side of the transmitter.

The purpose of the -90 degree and +90 degree measurements is to give a rudimentary indication of non-circular antenna variations on transmitter.

shanmugpriyan306, 2020-06-09:
Thanks @davidgyoung

shanmugpriyan306, 2020-06-15:
@davidgyoung , I am getting co-efficient A value as follows 9.741815429·10 ^-1
How can I use this value for "A co-efficient" 
[image]

davidgyoung, 2020-06-15:
9.741815429·10 ^-1 is the same as 0.9741815429.  Use that.