AltBeacon/android-beacon-library #984: Unable to stop advertising

norangebit, 2020-06-22:
### Expected behavior
```
beaconTransmitter.stopAdvertising();
```
This has to stop advertising.

### Actual behavior

The mobile will keep on advertising after the stopAdvertising method is called .

### Steps to reproduce this behavior

```
beaconTransmitter.startAdvertising(beacon)
beaconTransmitter.stopAdvertising()
```
### Mobile device model and OS version

- Xiaomi MI 5 - MIUI 10.2/Android 8.0.0
- Oneplus 3T - Oxygen 9.0.6/Android 9

### Android Beacon Library version

Android Beacon Library 2.17.1

I've checked #389, but I'm sure I'm using the same transmitter because the problem also occurs when the stop is called right after the start.

davidgyoung, 2020-06-22:
@norangebit please note tat that the call to start and stop are asynchronous.  If you call start and stop one right after the other in code, the result will be indeterminate.  If you really want to do this, you must wait for the callback from the longer startAdvertising method:

```
startAdvertising(Beacon beacon, android.bluetooth.le.AdvertiseCallback callback)
```

To troubleshoot can you please change your code to use the longer method with the callback and add log lines for when start is called, when stop is called and when the callbacks are called. 
 Please also log the instance of the beaconTransmitter in each of these cases so we can confirm it is the same instance.

norangebit, 2020-06-22:
@davidgyoung I wrote the code quickly without thinking it through too much. With the suggested changes, I solved it. Thank you.

closed 2020-06-22 by davidgyoung

jpunie, 2022-10-20:
I'm actually having a similar issue. 

When starting multiple broadcasts, it seems impossible to stop the earlier started broadcasts. 

The BluetoothLeAdvertiser stopAdvertising API does state that the callback used must be the same one as used in the startAdvertising. 

When looking through your BeaconTransmitter code, it will always use another instance. Why is this? 
Could this be related to the behaviour mentioned?