AltBeacon/android-beacon-library #985: Requesting Location Permission

deeplathia, 2020-06-23:
### Expected behavior
In the blog written on the github.io page of AltBeacon library, there's a section where you've mentioned changes needed to be made in requesting Location Permissions from Android 8, 9, 10. In that particular page when requesting ACCESS_BACKGROUND_LOCATION the only minimum SDK check kept is greater than Android M(26) but the min SDK for requesting ACCESS_BACKGROUND_LOCATION is Android Q(29). So the ACCESS_BACKGROUND_LOCATION should only be requested if the current SDK is 29 or above right?

Here's the link: https://altbeacon.github.io/android-beacon-library/requesting_permission.html

Please look into this.

davidgyoung, 2020-06-23:
In the link you shared it says it should only be quested if SDK is >= 29.  This is correct.  So what is wrong?

deeplathia, 2020-06-24:
Actually in the code snippet that you've shared when the target SDK >=29, The check when requesting the ACCESS_BACKGROUND_LOCATION is only if the SDK>=23 whereas there should also be a check if the SDK >=29 only then request for the ACCESS_BACKGROUND_LOCATION isn't it?