AltBeacon/android-beacon-library #993: removeRegion is called too late when using messenger

vladvladau, 2020-08-06:
### Expected behavior

1. disableMonitoring (regionBootstrap disable)
    - stopMonitoringBeaconsInRegion: scheduledScanJobs are enabled -> normal removeRegion call;
    - old region is removed;
2. enableForegroundService (disable scheduledScanJobs)
3. enableMonitoring (new regionBootstrap)
    - startMonitoringBeaconsInRegion: scheduledScanJobs are disabled -> messenger addRegion call;
    - new region is added;
4. disableMonitoring (regionBootstrap disable)
    - stopMonitoringBeaconsInRegion: scheduledScanJobs are disabled -> messenger removeRegion call;
    - old region is removed;
5. disableForegroundService (enable scheduledScanJobs)
6. enableMonitoring (new regionBootstrap)
    - startMonitoringBeaconsInRegion: scheduledScanJobs are enabled -> normal addRegion call;
    - new region is added;

### Actual behavior

1. disableMonitoring (regionBootstrap disable)
    - stopMonitoringBeaconsInRegion: scheduledScanJobs are enabled -> normal removeRegion call;
    - old region is removed;
2. enableForegroundService (disable scheduledScanJobs)
3. enableMonitoring (new regionBootstrap)
    - startMonitoringBeaconsInRegion: scheduledScanJobs are disabled -> messenger addRegion call;
    - new region is added;
4. disableMonitoring (regionBootstrap disable)
    - stopMonitoringBeaconsInRegion: scheduledScanJobs are disabled -> messenger removeRegion call;
    - **old region is not removed immediately, it is removed after point 6;**
5. disableForegroundService (enable scheduledScanJobs)
6. enableMonitoring (new regionBootstrap)
    - startMonitoringBeaconsInRegion: scheduledScanJobs are enabled -> normal addRegion call;
    - new region is added;

**Messenger removeRegion call comes later than normal addRegion call even if it's called before it.**

### Steps to reproduce this behavior

Call the sequence above;

**Relevant logs (steps 4, 5, 6):**
2020-08-06 03:35:51.824 D/BeaconService: BEFORE DISABLE FOREGROUND scheduled scan JOBS DISABLED
2020-08-06 03:35:51.825 D/BeaconManager: callback packageName: com.blesm
2020-08-06 03:35:51.825 D/BeaconManager: Unbinding
2020-08-06 03:35:51.828 D/BeaconManager: Before unbind, consumer count is 1
2020-08-06 03:35:51.828 D/BeaconManager: After unbind, consumer count is 0
2020-08-06 03:35:51.828 D/BeaconManager: This consumer is not bound.  Binding now: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@9762e5
2020-08-06 03:35:51.828 D/BeaconManager: Not starting beacon scanning service. Using scheduled jobs
2020-08-06 03:35:51.829 D/BeaconManager: callback packageName: com.blesm
2020-08-06 03:35:51.829 D/MonitoringStatus: saveMonitoringStatusIfOn()
2020-08-06 03:35:51.830 W/MonitoringStatus: ADDED REGION: 1
2020-08-06 03:35:51.837 D/MonitoringStatus: saveMonitoringStatusIfOn()
2020-08-06 03:35:51.841 D/BeaconManager: consumer count is now: 1
2020-08-06 03:35:51.841 D/AppStarter: Waiting for BeaconService connection
2020-08-06 03:35:51.841 D/BeaconService: AFTER DISABLE FOREGROUND scheduled scan JOBS ENABLED
2020-08-06 03:35:51.843 I/BeaconService: stop monitoring received
2020-08-06 03:35:51.843 D/BeaconService: stopMonitoring called
2020-08-06 03:35:51.843 D/MonitoringStatus: saveMonitoringStatusIfOn()
2020-08-06 03:35:51.843 W/MonitoringStatus: REMOVED REGION: 0
2020-08-06 03:35:51.843 D/BeaconService: Currently monitoring 0 regions.
2020-08-06 03:35:51.850 I/BeaconService: unbinding so destroying self
2020-08-06 03:35:51.850 E/BeaconService: onDestroy()
2020-08-06 03:35:51.851 I/BeaconService: onDestroy called.  stopping scanning

### Mobile device model and OS version
Huawei Mate 20 Pro, Android 10.1
### Android Beacon Library version
Latest

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

vladvladau, 2020-08-06:
I've managed to overcome this in my app with a dirty workaround.
I am now calling enableMonitoring (from step 6) after 100ms, in order to delay it to leave the messenger to process the request.
Let me know what you think, in order to achieve a more elegant solution.

davidgyoung, 2020-08-06:
Thanks for the update, Vlad.  I looked at this a bit earlier, and I think
it might happen because the monitoring regions are not persisted after the
service stops.  I need to look more.

vladvladau, 2020-08-10:
Hi David, did you managed to have a look on this? I still think this is a concurrency problem.

davidgyoung, 2020-08-10:
Sorry, @vladvladau, I have not had time to investigate a solution, and I have a very busy week, so I probably can't to so soon.

If you want to do some investigation yourself, here is my theory about what is happening:

I suspect the problem may be in this line:  https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconManager.java#L993

When you call RegionBoostrap disable, it calls BeaconManager stopMonitoringBeaconsInRegion, which calls the above method.  But if isAnyConsumerBound returns false, it just exits.

What should happen in this case, is it needs to persist the state of the monitored regions.

vladvladau, 2020-08-10:
Hi @davidgyoung. I've investigated your theory, and it's not the case.

This is being reached all the time https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/BeaconManager.java#L1013

All the functions are reached, but not in the correct order.
I will try to figure it out on my own, in the meantime let me know if you have any additional open points or you need further clarifications how to reproduce the issue.