AltBeacon/android-beacon-library #994: StartupBroadcastReceiver onReceive it's called repeatedly

vladvladau, 2020-08-10:
### Expected behavior

disableMonitoring (regionBootstrap disable) should stop any scheduled passive scan

### Actual behavior

If disableMonitoring (regionBootstrap disable) it's called immediately after receiving a passive scan intent, StartupBroadcastReceiver:onReceive will be called for setBackgroundScanPeriod seconds.

### Steps to reproduce this behavior

1. Run the app in background mode;
2. call regionBootstrap.disable(); in didEnterRegion();
3. Trigger the enter region state using HW filter (1st seen after 15 minutes);

**Relevant logs:**
2020-08-10 20:02:48.581 W/MessageQueue: Handler (android.os.Handler) {5c2e52} sending message to a Handler on a dead thread
    java.lang.IllegalStateException: Handler (android.os.Handler) {5c2e52} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:618)
        at android.os.Handler.enqueueMessage(Handler.java:771)
        at android.os.Handler.sendMessageAtTime(Handler.java:717)
        at android.os.Handler.sendMessageDelayed(Handler.java:687)
        at android.os.Handler.post(Handler.java:416)
        at org.altbeacon.beacon.service.scanner.CycledLeScanner.destroy(CycledLeScanner.java:271)
        at org.altbeacon.beacon.service.ScanJob.stopScanning(ScanJob.java:229)
        at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:212)
        at android.app.job.JobService$1.onStopJob(JobService.java:67)
        at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:117)
        at android.os.Handler.dispatchMessage(Handler.java:110)
        at android.os.Looper.loop(Looper.java:219)
        at android.app.ActivityThread.main(ActivityThread.java:8347)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1055)

Attached full logs.
[logs_intent_issue.txt](https://github.com/AltBeacon/android-beacon-library/files/5052244/logs_intent_issue.txt)

### Mobile device model and OS version
Huawei Mate 20 Pro, Android 10.1
### Android Beacon Library version
Latest

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

vladvladau, 2020-08-10:
My first finding question:

If I intentionally disableMonitoring (regionBootstrap disable), during an immediate scan, why is the ScanJob:onStopJob:startPassiveScanIfNeeded() if there are no consumers left?

davidgyoung, 2020-08-10:
If there are no consumers left, then there is no reason to do this.  These passive scans are intended to start when the main scan cycle stops (e.g. you are in the between scan period.)  It may be that there is simply a missing check to verify there are still consumers in this case.

vladvladau, 2020-08-10:
I would propose to check the `mScanState.getMonitoringStatus().regions().size() != 0`

Here: 
https://github.com/AltBeacon/android-beacon-library/blob/master/lib/src/main/java/org/altbeacon/beacon/service/ScanJob.java#L182

Let me know what you think

vladvladau, 2020-08-10:
Hi @davidgyoung, after applying this above proposed fix, I've faced the next issue if you disable the monitoring immediately after fresh started ScanJob:

ScanJob:stopScanning() is called twice triggered by RegionBootstrap:disable()

1. stopMonitoringBeaconsInRegion makes the monitored regions 0, leading to https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJob.java#L139
2. From https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJob.java#L208

Even if at point 1, getCycledScanner should be destroyed, when reaching https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJob.java#L223 it's evaluated as TRUE and returns `W/MessageQueue: Handler (android.os.Handler) {deb92d8} sending message to a Handler on a dead thread.

I've managed to overcome this with a dirty workaround, by delaying the execution of the RegionBootstrap:disable() with 30ms.
This way, https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJob.java#L114 returns TRUE, and stopScanning() is called only once when calling the disable function.

There has to a safety mechanism preventing stopScanning to be called twice.
Please let me know, when the time allows you, if you have any open points / recommendations regarding what I've described above.

`

davidgyoung, 2020-08-10:
@vladvladau 

The `stopScanning` method has a flag `mInitialized` that it sets to false.   Perhaps you can bail out of the method early if `mInitialized != true`?

And, yes, I agree with the proposed `mScanState.getMonitoringStatus().regions().size() != 0` change.

Thanks,
David

vladvladau, 2020-08-14:
Hi @davidgyoung. I tried to prevent the stopScanning to be called twice instead of making a mechanism inside it.
The issue is that the ScanJobs are canceled after we remove the region.
I would suggest to do it the other way around, but I think this is a more complex change in https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJobScheduler.java#L222

I've made a workaround by calling `ScanJobScheduler.getInstance().cancelSchedule(getApplicationContext());` before calling `disableMonitoring()`. This fixes my issue.

I will do a pull request for the above agreed change, but for the 2nd one, I prefer to receive your analysis when you have time, and let me know how should I approach it.

vladvladau, 2020-08-15:
I've also added here: https://github.com/AltBeacon/android-beacon-library/blob/c93cbc47c2d2cfc6ddf583094da91b472d7b1f8e/lib/src/main/java/org/altbeacon/beacon/service/ScanJobScheduler.java#L115

The following line: `mBackgroundScanResultQueue = new ArrayList<>();`

I was getting previous scan results after enabling monitoring.

davidgyoung, 2020-08-15:
@vladvladau  I appreciate your work on this.   I would very much like to get these issues resolved.   Per your comment above, "I prefer to receive your analysis when you have time, and let me know how should I approach it", are you specifically referring to the two calls to cancel the scan jobs, (both the immediate scan job and the scheduled scan job)?  

I looked at this code and I do see that if two such cancellation requests are made, and they result in the `ScanJob#onStopJob` method being called twice, this would stop the scan twice.  But I do not think this should happen unless both jobs are active at the same time (which should not be true -- only one should be active at a time).  My understanding if a cancel request is made for a Job Id that is not running, no call to `onStopJob` will be made and this problem should not happen.  Do you see otherwise?  Please clarify.

Apologies,  this kind of problem is very time consuming to track down (as I am sure you are discovering!) and I am heads down on a few other projects right now.  I did write all this code and understood it well when I wrote it about three years ago, but three years will fade knowledge a bit.

davidgyoung, 2021-08-13:
Checks for `mScanState.getMonitoringStatus().regions().size() != 0` added in 2.19 in autobind methods

closed 2021-08-13 by davidgyoung