AltBeacon/android-beacon-library #998: App cannot be started by RegionBootstrap from background

malibu1966, 2020-09-07:
Hi there.  I have been trying to use the 'start app in background' example and I am convinced there must be some bug.  I have taken this to the extent that I started a new 'drawer-based' example application and only applied the location permission and changes required to make background RegionBootstap monitoring work with a start of MainActivity when the beacon is detected.  I did .setDebug on BeaconManager and the beacon is being detected, also the didEnterRegion is being called in my Application class.

### Expected behavior

Application starts in foreground

### Actual behavior

09-07 11:34:30.359  1386  1544 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.PACKAGE_RESTARTED dat=package:com.example.beacontest flg=0x10 (has extras) } to android/com.kddi.android.CpaNotiReceiver

### Steps to reproduce this behavior

Install sample drawer based application
add beacon library
extend Application as per sample
modify Manitest as per sample
Add check/request for ACCESS_FINE_LOCATION and ACCESS_BACKGROUND_LOCATION

### Mobile device model and OS version
Samsung Note 9 with default launcher
Android version 10 kernel 4.9.186-18803994

### Android Beacon Library version
2.17.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2020-09-07:
@malibu1966, can you please try to reproduce this problem with the reference app?   https://github.com/AltBeacon/android-beacon-library-reference   If you can reproduce with that app as-is, or with a slightly modified version for which you can share a diff, we can help track down the issue in this forum.  If you want help debugging your custom app, the proper forum for help is stackoverflow.com.

closed 2020-09-07 by davidgyoung

malibu1966, 2020-09-08:
I tried the library reference.  The one modification I made was to scan for iBeacons.  I am getting the same error:

09-07 22:57:07.151  1424  1618 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.PACKAGE_RESTARTED dat=package:org.altbeacon.beaconreference flg=0x10 (has extras) } to android/com.kddi.android.CpaNotiReceiver

malibu1966, 2020-09-08:
Why was this closed?

davidgyoung, 2020-09-08:
@malibu1966, this was closed as of my initial comment because the steps to reproduce are about building a custom app, which makes it out of scope for this forum.  I do see your comment above about doing a test with the reference app.  Please open a new issue that describes the specific problem you see with the reference app and steps to reproduce  (e.g. launch app in foreground, turn on a beacon with identifier XXX, wait X minutes for a didEnerRegion callback, etc.)  An operating system log message is not necessarily unexpected behavior.  What is the functional behavior that you see that is unexpected?