AltBeacon/android-beacon-library #1003: Access Advertising Flags (such as, BR/EDR) using AltBeacon Lib?

yong3liu2, 2020-09-25:
Hello,

Using Altbeacon lib, I can see the beacons that I am looking for in public void didEnterRegion(Region region) { ...}

But I also would like to read AD Flags such as following, for example: 
02 # Number of bytes that follow in first AD structure 
01 # Flags AD type 
1A # Flags value 0x1A = 000011010 
bit 0 (OFF) LE Limited Discoverable Mode 
bit 1 (ON) LE General Discoverable Mode 
bit 2 (OFF) BR/EDR Not Supported 
bit 3 (ON) Simultaneous LE and BR/EDR to Same Device Capable (controller) 
bit 4 (ON) Simultaneous LE and BR/EDR to Same Device Capable (Host)

Is there an API in Altbeacon lib to read the Flags?  I do see Android has something like: https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getAdvertiseFlags()

Just not finding anything like it in Altbeacon lib yet.  It would be nice to have it.

Thanks, 
Yong

davidgyoung, 2020-09-25:
The library currently ignores flags AD types, so this is currently not possible.    It would be possible to add support for parsing this information.  If you are interested in contributing to the library, I could give you some tips on how to add such a feature.

davidgyoung, 2024-01-14:
Upon further review, this is not supported by Android APIs.

closed 2024-01-14 by davidgyoung