AltBeacon/android-beacon-library #1004: Mass Crashes org.altbeacon.beacon.service.ScanJob -> java.lang.RuntimeException: android.os.DeadSystemException

positive-energi, 2020-10-06:
We are getting thousands of crashes 

App is in background 

LG & Motorola OS 10

Library version 2.16.3

Fatal Exception: java.lang.RuntimeException
Unable to unbind to service org.altbeacon.beacon.service.ScanJob@fe37965 with Intent { cmp=com.dollargeneral.android/org.altbeacon.beacon.service.ScanJob }: java.lang.RuntimeException: android.os.DeadSystemException
android.app.ActivityThread.handleUnbindService (ActivityThread.java:4139)
android.app.ActivityThread.access$1700 (ActivityThread.java:226)
android.app.ActivityThread$H.handleMessage (ActivityThread.java:1924)
android.os.Handler.dispatchMessage (Handler.java:107)
android.os.Looper.loop (Looper.java:214)
android.app.ActivityThread.main (ActivityThread.java:7615)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:492)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:964)

Caused by java.lang.RuntimeException: android.os.DeadSystemException
       at android.app.ActivityThread.handleUnbindService(ActivityThread.java:4133)
       at android.app.ActivityThread.access$1700(ActivityThread.java:226)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1924)
       at android.os.Handler.dispatchMessage(Handler.java:107)
       at android.os.Looper.loop(Looper.java:214)
       at android.app.ActivityThread.main(ActivityThread.java:7615)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

Caused by android.os.DeadSystemException
       at android.app.ActivityThread.handleUnbindService(ActivityThread.java:4133)
       at android.app.ActivityThread.access$1700(ActivityThread.java:226)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1924)
       at android.os.Handler.dispatchMessage(Handler.java:107)
       at android.os.Looper.loop(Looper.java:214)
       at android.app.ActivityThread.main(ActivityThread.java:7615)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

emilewakim96, 2020-10-06:
Facing same issue!

davidgyoung, 2020-10-06:
The core cause of this exception is a DeadSystemException:

The core Android system has died and is going through a runtime restart.
All running apps will be promptly killed.

See here for details:
https://developer.android.com/reference/android/os/DeadSystemException

The root cause is a crash or restart of the operating system.  This likely
has absolutely nothing to do with your app.  The stack trace is just an
indication of how the OS cleans up your running app when this happens.  You
cannot fix this problem because you cannot prevent the OS from crashing
when your app happens to be running.  The crash is not your fault and is
outside your control.

If you are seeing this in some automated crash reporting system, then you
should simply ignore it.  If you need some kind of justification for doing
so, perhaps referencing this response would help.

closed 2020-10-06 by davidgyoung